Slamtec/rplidar_ros #1: RPLIDAR A3: SIGSEGV on Raspberri Pi

qnux, 2018-08-09:
Hi,

The _rplidarNode_ is working very fine on Ubuntu x86 with my RPLIDAR A3. (Linux xxxxxx 4.15.0-30-generic #32~16.04.1-Ubuntu SMP Thu Jul 26 20:25:39 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux)

 * /rosdistro: kinetic
 * /rosversion: 1.12.13

```
rplidar_ros$ git log -1
commit 18427915161513d97f808b18ccc1a35d53c276a1
Author: kintzhao <kint.zhao@slamtec.com>
Date:   Thu Jul 19 04:08:24 2018 -0700

    [release] rplidar_ros release 1.7.0
```

Using the **same** ROS version and _rplidar_ros_ version on a Raspberri Pi3 with this kernel : Linux xxxxxx 4.9.80-v7+ #1098 SMP Fri Mar 9 19:11:42 GMT 2018 armv7l armv7l armv7l GNU/Linux

When I launch rplidar_a3.launch, the sensor starts spinning and display its serial number (indicating that the comm seems OK), but then quickly ends with a SIGSEGV:

```
[ INFO] [1533846640.975505892]: RPLIDAR running on ROS package rplidar_ros. SDK Version:1.7.0
RPLIDAR S/N: xxxx
[ INFO] [1533846641.483938270]: Firmware Ver: 1.24
[ INFO] [1533846641.484141603]: Hardware Rev: 6
[ INFO] [1533846641.486406081]: RPLidar health status : 0
[ INFO] [1533846642.140504057]: current scan mode: Sensitivity, max_distance: 25.0 m, Point number: 15.9K , angle_compensate: 4
[rplidarNode-1] process has died [pid 3943, exit code -11, cmd /home/ubuntu/catkin_ws/devel/lib/rplidar_ros/rplidarNode __name:=rplidarNode __log:=/home/xxxxx/.ros/log/759e04f8-d0dc-11e5-8f11-53a2f0703914/rplidarNode-1.log].
```

Going further in the diagnosis, I launch with gdb, that indicates me the segfault occurs in `rplidar_ros/src/node.cpp:352` which is the call to `ros::spinOnce();`

Thanks you for the help.
I am available to provide further details if needed.

Tejas-Shastha, 2018-08-14:
Facing a similar issue with my new A3. However, the node does not crash immediately, rather only when I bring my hand close to it. Strangely enough, this only happens with my hand and not my colleague's!
What is interesting however, is that on one machine with a previous version of the binary (1.6.0 per package.xml) it only crashed once but worked well most of the time, even with my hand close. However, replicating the same version from source on a different machine still produces the crash.

kintzhao, 2018-08-30:
@qnux   https://github.com/robopeak/rplidar_ros/blob/master/src/node.cpp#L270 you can try to change the code by an const angle_compensate_multiple ;
//angle_compensate_multiple = (int)(1000*1000/current_scan_mode.us_per_sample/10.0/360.0);
angle_compensate_multiple = 1;

kintzhao, 2018-08-31:
@qnux  what is  the reason,  How to solved it . The solution is fine for someone who may hit

kintzhao, 2018-08-31:
@Tejas-Shastha   I test it is ok when lidar closed to obs on PC .   if you work at Raspberri Pi3,   you can try decrease the angle_compensate_multiple  ref  https://github.com/Slamtec/rplidar_ros/issues/1#issuecomment-417173666

Timple, 2018-08-31:
Solution works also on a Olimex Lime2 (another arm board). So issue seems arm specific.

matth-c3, 2018-12-31:
Hello @kintzhao , How much point are you getting ? [Here](https://webcache.googleusercontent.com/search?q=cache:mSWyS7etIQcJ:https://diyrobocars.com/2018/08/04/lidar-slam-without-ros-on-cheap-computing-boards/+&cd=7&hl=fr&ct=clnk&gl=fr) they mention 75 points at one update/sec, with lots of artifacts. 

Thanks by advance

salihmarangoz, 2019-02-16:
Apply this patch: #10 
This solves SIGSEGV problem.

salihmarangoz, 2019-02-20:
> Apply this patch: #10
> This solves SIGSEGV problem.

Also don't forget to add "scan_mode" parameter and set to "Boost". You will get more points :)

deb0ch, 2019-03-25:
I had a similar segfault today on the same on a Raspi and RPLidar A3, in Stability mode, consistently a few seconds after the node is launched.

It got fixed with a patch similar to the one in https://github.com/Slamtec/rplidar_ros/pull/10

RedEyeDragon, 2020-07-08:
I bought the RPLiDAR A3M1 and am trying to get this sensor to run. I have tried to get it running on a Raspberry PI 4 running Ubuntu Buster with no luck. I have followed any and all instructions including a very long tutorial to install ros_kinetic. After getting ros installed I tried to follow this tutorial. https://medium.com/robotics-with-ros/installing-the-rplidar-lidar-sensor-on-the-raspberry-pi-32047cde9588 but the error I get when running catkin_make is : pi@RedEyeDragon:~/rplidar_ws/src $ catkin_make
Base path: /home/pi/rplidar_ws/src
The specified base path "/home/pi/rplidar_ws/src" contains a package but "catkin_make" must be invoked in the root of workspace
pi@RedEyeDragon:~/rplidar_ws/src 
Any help?

deb0ch, 2020-07-09:
What the error message is telling you is that you have to execute `catkin_make` from `~/rplidar_ws`, most likely you weren't there when you ran that command.

Btw, you might want to surround your terminal outputs with `` ``` `` on the lines above and under your pasted output to avoid Github's Markdown messing with it :

````
```
Your terminal output here
```
````