Slamtec/rplidar_ros #3: udev rule causes occasional failure on launch

Achllle, 2018-10-04:
I occasionally get an error launching my A2 indicating that ROS can't establish a connection with it, the same error as when the lidar isn't plugged in. The problem is resolved when the USB is unplugged and replugged. This made me suspect that the problem is due to the udev rule in the scripts folder and indeed, when I directly specify the portname using the ttyUSB* name, I never get the error. I tested this by repeatedly launching through the symlink 'rplidar' and continuously failing and the specifying the direct portname instead and it immediately working.

giuseppe-chiari, 2019-11-13:
I have the same problem, how did you manage to solve it? How did you create the udev file?

Achllle, 2019-11-13:
Unfortunately I'm not 100% sure how I resolved this. I do remember having another usb device with a udev rule that happened to be conflicting with the rplidar. Try removing the other udev rule and device and see if the issue persists after reloading the rules.

Some other things that might help:
* make sure you're supplying enough power to your usb devices, especially if you have multiple connected to the same port. Consider getting a powered usb hub in that case. 
* My udev rule for the A2 is the same as the one provided, but I assigned an importance to it by naming it `97-rplidar.rules`.

HomeLH, 2019-11-14:
same problem.

> nano@nano-desktop:~/catkin_ws$ roslaunch rplidar_ros rplidar.launch
... logging to /home/nano/.ros/log/407d6c14-06bb-11ea-b433-00e04c680001/roslaunch-nano-desktop-11156.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nano-desktop:39699/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /rplidarNode/angle_compensate: True
 * /rplidarNode/frame_id: laser
 * /rplidarNode/inverted: False
 * /rplidarNode/serial_baudrate: 115200
 * /rplidarNode/serial_port: /dev/ttyUSB0

NODES
  /
    rplidarNode (rplidar_ros/rplidarNode)

ROS_MASTER_URI=http://localhost:11311

process[rplidarNode-1]: started with pid [11171]
[ INFO] [1573723888.143559397]: RPLIDAR running on ROS package rplidar_ros. SDK Version:1.10.0
[ERROR] [1573723890.651973028]: Error, operation time out. RESULT_OPERATION_TIMEOUT!
[rplidarNode-1] process has died [pid 11171, exit code 255, cmd /home/nano/catkin_ws/devel/lib/rplidar_ros/rplidarNode __name:=rplidarNode __log:=/home/nano/.ros/log/407d6c14-06bb-11ea-b433-00e04c680001/rplidarNode-1.log].
log file: /home/nano/.ros/log/407d6c14-06bb-11ea-b433-00e04c680001/rplidarNode-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

giuseppe-chiari, 2019-11-14:
@Achllle yeah I think the problem was that two usb connected at the same time create problems in udev rules and usb port assignment, I've created a rplidar udev rule (before there wasn't) and it seems to have solved the issue. Now when the system boots the lidar is activated and starts working.

giuseppe-chiari, 2019-11-14:
@HomeLH I had the exact same output as you do and creating a udev rule for the lidar and the assigning the symlink to the port to the corresponding parameters in node.cpp and rplidar_a3.launch solved the issue (for now at least) and to add the new rule to /etc/udev/rules.d for Ubuntu/Fedora like systems. Remember to refresh and restart the udev service after adding the new rule, unfortunatly I don't remember the exact command

HomeLH, 2019-11-15:
@giuseppe-chiari Thank you. I got a wrong defaut baudrate bug with my A2M7. I hesitate it only supports 256000 baudrate while the rplidar_ros default is 115200.
Details at https://github.com/Slamtec/rplidar_ros/issues/5#issuecomment-554166786

Shelfcol, 2020-09-25:
I change 25600 to 115200, the rplidar_a3 worked

deyouslamtec, 2023-07-13:
summary：
·Supplying enough power
·Use the correct port and Baud

closed 2023-07-13 by deyouslamtec