Slamtec/rplidar_ros #4: ROS2 Port

allenh1, 2018-10-26:
Hey!

I ported `rplidar_ros` to `ros2`, and would prefer to move my work over here so it can be released, and would like to start a conversation about this so that we can do this efficiently.

My [fork](https://github.com/allenh1/rplidar_ros) has the port on the `ros2` branch. It is my opinion that it would be best to move that branch here.

Anxious to hear your thoughts!

tony-slamtec, 2018-10-30:
Hi. It's great to here this. Would you please paste the url of your repo in this issue? We will do some research, and maybe we can merge your code with a pull request after that.

Thanks again

allenh1, 2018-10-30:
@tony-slamtec great!

https://github.com/allenh1/rplidar_ros

brean, 2019-10-20:
Hey, it's nearly a year, so has this been merged? I can not find a pull request...

allenh1, 2019-10-21:
> I can not find a pull request...

I have not filed a pull request.

@tony-slamtec is there a branch I can file it to?

Bert-Tec, 2020-04-24:
any news here?

allenh1, 2020-04-24:
> any news here?

Nope.

@tony-slamtec I would like to release the ROS 2 driver, and I'll support it (that way it can be `sudo apt install ros-eloquent-rplidar-ros` installed) if you do not have the resources or desire to maintain a ROS 2 version.

I'd like to make sure that I'm not stepping on any toes here. If you would like to support a ROS 2 version in the future, I won't release it and will just continue to carry on my fork. Just let me know what your plan is, and I'll act accordingly.

VotionSam, 2020-04-25:
Just got my hands on an A3.   Really appreciates Allen's work.   But as a commercial product, I would expect ROS2 support to come from official repo.

Bert-Tec, 2020-04-25:
@VotionSam I'm with you. I can't understand why such a bigger company which sells a lot of the lidar's is not able to make a ros2 Port. I looked into the code and it's max. about half a day work to make a ros2 version.

NYP-RoboticsSG, 2020-10-01:
Anyone use it on A3, on ROS2 foxy, I kept getting time out problem

allenh1, 2020-10-02:
> Anyone use it on A3, on ROS2 foxy, I kept getting time out problem

I don't have an A3, but would be glad to help you figure it out. Please file an issue on my fork and we'll get on it!

gregorthebigmac, 2020-12-10:
Is this still not working on ROS2? I'm running Foxy, and whenever I run ```$ ros2 launch rplidar_ros view_rplidar.launch.py``` it bails with an error, saying ```executable 'rplidarNodeClient' not found on the libexec directory /opt/ros/foxy/lib/rplidar_ros``` and when I checked, it was correct. That executable is missing.

allenh1, 2020-12-10:
> Is this still not working on ROS2?

@gregorthebigmac I'm a little confused. Where did the version of the driver you installed come from?

If you're in foxy, you can actually `apt-get` install the driver built from my fork.

```
executable 'rplidarNodeClient' not found on the libexec directory /opt/ros/foxy/lib/rplidar_ros
```

The reason I ask is that the executable `rplidarNodeClient` is _not_ built by my fork, and is also not mentioned in any launchfiles (at least to my knowledge).

The launch file you mention includes `plidar.launch.py`, which uses the `rplidar_composition` executable (see [here](https://github.com/allenh1/rplidar_ros/blob/a23297b938edf0f795975794a9816ad6f3520b6d/launch/rplidar.launch.py#L10)).

Please feel free to follow up with an issue on the fork. Cheers.

gregorthebigmac, 2020-12-10:
Oh, sorry. I don't know if I installed your fork, I just grabbed the rplidar binary from apt using ```$ sudo apt-get install ros-foxy-rplidar-ros``` and tried launching the built-in packages. Is your fork necessary to get it running?

allenh1, 2020-12-10:
>  I just grabbed the rplidar binary from apt 

As of foxy, that is from my fork of the code. Could be something went wrong with the release. I'll give it a go later and see what happens.

gregorthebigmac, 2020-12-10:
If you need additional info from me, let me know. It was a pretty vanilla setup, as far as I can tell. I'm on Ubuntu 20.04, added my user to the dialout group, installed the package, plugged the rplidar in, and ran some of the built-in packages, and everything gave me errors. I pasted a dump of my terminal output when I ran it here https://gist.github.com/gregorthebigmac/96b5fc4c08988b039cd3e81f50812fc6

bartekpacia, 2021-01-19:
I'd love to see @allenh1's fork merged. @tony-slamtec, can you provide any info regarding ROS2 support?

zmontross, 2021-02-20:
Starting from literally nothing I was able to get @allenh1's fork working on Linux Mint 20 (which is effectively Ubuntu 20 package-wise).
For anyone also starting from scratch here are the steps I took:
- Installed ROS2 based on [these instructions](https://index.ros.org/doc/ros2/Installation/Crystal/Linux-Install-Binary/) (swapping 'crystal' for 'foxy' for ROS stuff, and using 'focal' for Ubuntu stuff) (also had to change some things to 'python3-').
- Sourced the ROS2 setup bash script: 'source $ROS2_WS/ros2-linux/setup.bash'
- Cloned @allenh1's fork, creating an rplidar_ros directory: 'cd $ROS2_WS/; git clone $ALLENH1_GIT_URL'
- Used Colcon to build the rplidar_ros package: 'cd rplidar_ros; colcon build --symlink-install'
- SIDENOTE: I had to install some OpenSSL packages based on output from Colcon: 'sudo apt install openssl libssl-dev'
- My user needed to be added to the dialout group: 'sudo usermod -aG dialout $(whoami)'
- Log-out of user account so that user group change is accepted. Log back into account, and navigate to ROS2 workspace: 'cd $ROS2_WS'
- Source the ROS2 Foxy (the version I installed) setup script: 'source ros2-linux/setup.bash'.
- Source the rplidar_ros setup script: 'source rplidar_ros/install/setup.bash'
- Run the launch python script (in my case the RViz visualizer) as per @gregorthebigmac above: 'ros2 launch rplidar_ros view_rplidar.launch.py'

This was my first time working with ROS or ROS2, and far too many how-to's refer to 'catkin_make'. [This article on TheConstructSim.com](https://www.theconstructsim.com/ros2-vs-ros1/) spelled out the differences very plainly.

Thanks for creating your fork @allenh1! It worked like a charm once I figured out what I was looking for. It's a shame we haven't heard from @tony-slamtec in so long.

allenh1, 2021-02-20:
@zmontross nice work! Thanks for all the info!

If you want to add this to the README on the fork, I'd happily accept a PR from you!

enricosutera, 2022-07-27:
I still am confused about how these ros drivers are handled by Slamtec.
If I am not wrong, the debian packages are still released by @allenh1 and for sure the [Slamtec/sllidar_ros2](https://github.com/Slamtec/sllidar_ros2) cannot be considered the new official ros2 repo. 

Also, why the repo changed name to sllidar_ros2?