Slamtec/rplidar_ros #5: Operation time out lidar

timdori, 2018-10-29:
hey guys,

i got a little issue while i was tried to run lidar. I followed the instructions on the website below.

http://wiki.ros.org/rplidar

But wehen i:
rosrun rplidar_ros rplidarNode
or
roslaunch rplidar_ros view_rplidar.launch

I get this Error: Error, operation time out. RESULT_OPERATION_TIMEOUT!

I tried the same with another PC and all was fine...
Could anyone please help me? Im out of ideas...
Thank you very much!

ameykasar, 2018-10-30:
i am getting the same error on my Raspberry Pi 3 B.
Detailed description of error [here](https://answers.ros.org/question/307109/result_operation_timeout-error-on-raspberry-pi-3-b-when-interfacing-rplidar-a1-with-rplidar_ros-package/)

HomeLH, 2019-11-14:
same problem. The port and permission are correct. Please check.
https://github.com/Slamtec/rplidar_ros/issues/3#issuecomment-553804434

ameykasar, 2019-11-14:
It was due to dodgy USB cables in my case.
I was powering my RPi with a USB and it wasn't giving the required power to RPi (Which therefore wasn't giving the required power to LIDAR). Tried it with a AC power adapter. And it was solved.
Would also suggest you to check the USB you are using to connect LIDAR to RPi.

HomeLH, 2019-11-15:
> It was due to dodgy USB cables in my case.
> I was powering my RPi with a USB and it wasn't giving the required power to RPi (Which therefore wasn't giving the required power to LIDAR). To spot this, look for the indicator LEDs on RPi before and after running LIDAR. Tried to power with an AC power adapter. And it was solved.
> Would also suggest you to check the USB you are using to connect LIDAR to RPi.

Thank you for your reply. I got a baudrate error. My device type is A2M7. I wonder it just support baudrate 25600 with hardware, but the default baudrate for A1/A2 is 115200. When I change the default baudrate to 25600, the motor runs and everything works well.

AlessandroMelino, 2020-04-14:
Hello everyone, I have the same issue.

I am using a RPLidar A1M8 to generate obstacles using move_base and costmap_2d. The problem arrives when I launch rplidar node with the whole system .launch; you can see the include where I call the .launch file of rplidar:

  <include file="$(find rplidar_ros)/launch/rplidar.launch" />

The next error appears on terminal when I launch it:

 [ERROR] [1585914589.956502460]: Error, operation time out. RESULT_OPERATION_TIMEOUT! 
[rplidarNode-20] process has died [pid 14017, exit code 255, cmd /home/sara/catkin_ws/devel/lib/rplidar_ros/rplidarNode __name:=rplidarNode __log:=/home/sara/.ros/log/35a653a0-75a1-11ea-83b0-701ce7079383/rplidarNode-20.log].
log file: /home/sara/.ros/log/35a653a0-75a1-11ea-83b0-701ce7079383/rplidarNode-20*.log

I tried to read the .log file, but it doesn't appear on that directory.

By the way, if I launch first the whole system in one terminal and then I launch the rplidar node using other terminal, it works and I can see rplidar working and generating obstacles on rviz.

What can I do to solve the problem and launch rplidar whit the system together?

I have checked permission and ports and it is everything ok.

Thanks in advance.

garyrightbot, 2020-10-13:
@AlessandroMelino I am facing the same issue. Were you able to solve this?

AlessandroMelino, 2020-10-13:
Hello @garyrightbot .

Yes, it finally worked for me.

I launch the Rplidar node with this code on the general .launch file:

`  <include file="$(find sara_2dnav)/launch/amcl_lidar/rplidar.launch" />
`

Then the rplidar.launch file is as following:

```
<launch>
  <node name="rplidarNode"          pkg="rplidar_ros"  type="rplidarNode" output="screen">
  <param name="serial_port"         type="string" value="/dev/ttyUSB1"/>
  <param name="serial_baudrate"     type="int"    value="115200"/><!--A1/A2 -->
  <!--param name="serial_baudrate"     type="int"    value="256000"--><!--A3 -->
  <param name="frame_id"            type="string" value="laser"/>
  <param name="inverted"            type="bool"   value="false"/>
  <param name="angle_compensate"    type="bool"   value="true"/>
  </node>
</launch>
```

Keep in mind the serial port where lidar is connected, in my case ttyUSB1, you can check it connecting it and executing `dmesg` in terminal.

I hope that this could help you.

Best regards.
Alessandro

duccoiatvt1108, 2021-05-09:
@AlessandroMelino  i dont know about ?

> <include file="$(find sara_2dnav)/launch/amcl_lidar/rplidar.launch" />

duccoiatvt1108, 2021-05-09:
@AlessandroMelino  i dont know about `<include file="$(find sara_2dnav)/launch/amcl_lidar/rplidar.launch" />`?

AlessandroMelino, 2021-05-10:
> @AlessandroMelino i dont know about `<include file="$(find sara_2dnav)/launch/amcl_lidar/rplidar.launch" />`?
> when my case is ttyUSB0

Yes, it was because I have another device on ttyUSB0.

hamsterasu, 2021-06-10:
I have the same problem and it turned out that the baudrate in the launch file was wrong for me. There was a physical switch in the adaptor to change the baudrate between 115200 and 256000. In my case, I forgot that I switched it to 256000 and so when I ran rplidar.launch with the default parameters, it did not work.

MikeIrlbeck, 2021-10-30:
I tried all the solutions above and many more but they all did not work for me.  Thus I deemed it necessary to share what I learned in case anyone else may find themselves in my situation.

I had this same exact error when trying to run with ROS and solely the SDK.  My rplidar s1 worked on my friend's computer but not mine.  I followed all the installation steps and received the "Error, operation time out. RESULT_OPERATION_TIMEOUT!" after trying roslaunch rplidar_ros view_rplidar_s1.launch.

SOLUTION for me (not presented above): do NOT use your computer for the power supply, instead use a phone wall adapter (5V 1A).  

Here is the troubleshooting [checklist](https://blog.csdn.net/zkk9527/article/details/109177275) I found:
1. check that baud rate is consistent with code 
2. check if there is a problem with the power supply; at least 5V, 1.5 A ("Insufficient power supply usually results in operation time out error") (my note - 5V and 1 A did the trick)
3. Third, check whether the USB port of the radar is ttyUSB0 in the launch file
    1. ls -l /dev/ttyUSB*       
    2. port permissions: sudo chmod 0777 /dev/ttyUSB0 (0666 also works to my knowledge)
4. Lidar may be broken  
    1. Check by running with just the sdk
    2. Preferably try running on windows (.exe)

jcocovich, 2022-11-12:
Yeah. My Canakit wall power supply doesn't have a cord long enough for my turtlebot3. LOLOL.
Bench power supply set at 5.1v and 3.5A fed through the USB-C port on my RPI4. Same time out error.
There appears to be no way to use the Slamtec RPLidar A1M8 through the USB converter to my PI. Just sayin'

A1exan10er, 2023-01-27:
> I tried all the solutions above and many more but they all did not work for me. Thus I deemed it necessary to share what I learned in case anyone else may find themselves in my situation.
>
> I had [...]

The 3. method saved me. Thank you with lots of love!!! ;)

jcocovich, 2023-01-28:
My response to your checklist:
1. Everything running is out of the box code with no modifications.
2. As mentioned, my Canakit power supply doesn't have a long enough cord and, besides, it does not work either plugged directly into the USB power port on the RPI4.
3. Everything is running out of the box code with no modifications.
4. THAT WOULD BE MY CONCLUSION. Slamtec just says I'm not using it properly and RTFM.

AGAIN, just sayin'.

AbdulrahmanAlGhofaily, 2023-02-15:
For me, I solved the problem by simply switching the `serial_buadreate` from 115200 to 256000, and `angle_compensate` to False. Both on rplidar.launch.py file.
Running the Lidar on Raspberry pi 4, Ubuntu 22.04, and ROS2 of course. RPLiDAR A2M12.
Recommend you yo go through the checklist the brother made up there.

jayren17, 2023-06-08:
@jcocovich Hi there. It's been a while, but did you manage to find a solution to your issue? Encountering the same problem with my RPI4 setup to LiDAR and the fixes above don't seem to work.

AbdulrahmanAlGhofaily, 2023-06-08:
@jayren17 Hey there, I had the same issue before, I was working on the A2m12 rplidar together with the RPI4, I used Power Bank to power my RPI4, through which I power the A2m12. Honestly I forgot the details about the operating system I used and the setup and even how did I download the libraries, but this is my code through which everything worked fine, hope it helps you at least get your A2m12 works. (Note: Some of the functions in the code are related to the pins to power some motors, it was a vehicle that navigate using a lidar project. Just ignore it.)
https://github.com/AbdulrahmanAlGhofaily/vehicle_controller/blob/master/vehicle_controller/main_comp.py

AbdulrahmanAlGhofaily, 2023-06-08:
> @jayren17 Hey there, I had the same issue before, I was working on the A2m12 rplidar together with the RPI4, I used Power Bank to power my RPI4, through which I power the A2m12. Honestly I forgot [...]

Here is the report with more details about the project (Autonomous service vehicle using LiDAR technology). I went in more details about the OS that is used. You could ignore everything and jump straight to the third section which talks about the parts that I used in the project and the OS that I've used.
https://docs.google.com/document/d/e/2PACX-1vQo91XuXmmS-7n4A_ce96Lvt7xmF4NwyiEO5pkvxccYB02evkL2AKaKtiJr6zG6jl0cr47hvxp5bFUC/pub