Slamtec/rplidar_ros #12: Much more noise with RPLidar A3 than A2 under sunlight

deb0ch, 2019-03-26:
I just tried my new RPLidar A3 in my backyard, and even in Stability mode the signal was drastically noisier than with the RPLidar A2 under the exact same conditions.

I ran them both using `rplidar_ros` on a Raspberry Pi 3 Model B, the A2 using the default `rplidar.launch`, so in `Express` mode, and the A3 using `rplidar_a3.launch`, changing the mode to `Stability`.

It was clear the the source of the noise was the sun: at the moment the clouds were occulting the sun the signal improved drastically, going back to noise when the sun shined again. The RPLidar A3 wasn't even able to see a big grey garbage can that was 3m in front of it under sunlight but it could with the clouds. A lot of points were falsely detected close to the lidar, at about 0.3m. The A2 worked perfectly under both conditions.

Here is an illustration of the problem with the RPLidar A3:

[image]

On this curve, the X axis is time in seconds and the Y axis the distance in meters to an obstacle. To get it, I discarded all the points except in a 90º field of view in front of the lidar, and then among those points, selected the one closest to the lidar and applied a `cos` of the angle to get the forward distance. I then moved the lidar in front of a big wooden plank.

You can see an area where the signal gets cristal clear: when clouds passed in front of the sun. 

For comparison, here is the RPLidar A2 under the same conditions:

[image]

The signal is a lot better under the same sunlight, which I did not expect.

Robustness to sunlight is critical in my application. **What kind of settings should I use for the RPLidar A3, that would make it resist sunlight better than `Stability` ?**

milan-r-shah, 2020-10-12:
Hi @deb0ch ,
Could you please let me know how did you figure out that you should update the `value` to `stability` from `sensitivity` for the `scan_mode` parameter? I couldn't find that anywhere, not even in the source code. On the other hand, the website says

> RPLIDRA A3 supports to work under two modes alternatively: enhanced mode and outdoor mode.

Do they have enough documentation for the ROS package? I couldn't find that either!

It's surprising that it's been 1.5 years since you created this issue and they haven't responded yet! Anyway, were you able to figure this out?

Thank you so much in advance!

milan-r-shah, 2020-10-12:
One more thing is, in **rplidar_ar.launch**, if I keep `scan_mode` to `Sensitivity` and launch **view_rplidar_a3.launch** then in the terminal, I gets this:
```
current scan mode: Sensitivity, max_distance: 25.0 m, Point number: 15.9K , angle_compensate: 4
```

On the other hand, if I change `scan_mode` to `Stability` then it gives this:
```
current scan mode: Stability, max_distance: 25.0 m, Point number: 10.0K , angle_compensate: 2
```

So, now I'm even more confused because, in `Stability` mode, I'm getting fewer points compared to `Sensitivity` mode. So, which `scan_mode` should I use for the outdoor application?