Slamtec/rplidar_ros #13: A2 too much inf even indoors at midnight without ceramic tile

HaoQChen, 2019-04-02:
inf data may up to 1/3
What can I do with this? Should I support power independently beside USB? Or change some setting. I use the default setting.

HaoQChen, 2019-04-02:
using Standard scan mode is getting  worse. 1/2 inf

HaoQChen, 2019-04-02:
[image]
inf count 250+ with total 360

HaoQChen, 2019-04-02:
I have supported power independently btw. I want to curse.

anakin0620, 2022-05-23:
> Could you describe the sounding? If there are a lot of very dark surface, it may cause this issue. The reflectivity of those surface are pretty low which will cause very weak laser back to the receiver inside of RPLiDAR.