Slamtec/rplidar_ros #21: No Parameter for Specifying Motor Speed

erush91, 2019-12-06:
The RPLIDAR runs at the default speed, and there is no way to specifying the motor speed.

Motivation:
In robotics applications, users may want to run at a faster rate to have better collision avoidance. Or perhaps users have limited computation power and would like to run at a slower rate.

Facts:
The RPLIDAR S1 can run between 5 - 15 Hz, but it's speed is permanently set to 10 Hz in the ROS node.
The RPLIDAR A3 can run between 10 - 20 Hz, but it's speed is permanently set to 10 Hz in the ROS node.

Value Proposal:
The end user would really benefit from being able to vary the motor speed based on their application needs. And since the hardware and SDK are capable of this, the ROS node really should be as well.

Thanks!

kerliang0801, 2020-01-02:
I second this proposal. It seems very reasonable, in fact, it should be fairly easy as well! :)

lukelu0520, 2020-07-22:
push it too. It could be adjusted in windows (using slamtec robot studio), but can not tune it in ROS, doesn't make sense.

jash101, 2021-04-16:
@tony-slamtec this addition can be really helpful. Are you guys planning to work on it?

Nikhil619, 2023-12-05:
> push it too. It could be adjusted in windows (using slamtec robot studio), but can not tune it in ROS. The most closest solution I found is here: https://github.com/davidbsp/rplidar_ros

Looks good but did it work by mentioning the pwm value as required in the node.cpp file ? 
Have you tested it?

Thanks in advance