Slamtec/rplidar_ros #33: Working on Noetic using Ubuntu 20.04 ?

lukelu0520, 2020-07-31:
Hi rplidar_ros friends,

We were wondering if anyone run rplidar_ros on Ubuntu 20.04 (w/ ROS Noetic Ninjemys). Is it working as stable as in Melodic using Ubuntu 18.04 or Kinetic using Ubuntu 16.04 ?

Best,
-- Luke

AndreV84, 2020-09-09:
also if it works with ros2 foxy? on 20.04?

adamkrawczyk, 2020-09-24:
It works the same way as in melodic, also works with ros2 but I got a little malformed shape

AndreV84, 2020-09-24:
Thank you for following up!

simonbogh, 2020-10-22:
Will there be a release to Noetic so it can be installed from binaries?

deyouslamtec, 2023-05-12:
> Dear rplidar_ros friends,
> 
> We were wondering if anyone run rplidar_ros on Ubuntu 20.04 (w/ ROS Noetic Ninjemys). Is it working as stable as in Melodic using Ubuntu 18.04 or Kinetic using Ubuntu 16.04 ?
> 
> Best, -- Luke

Yes

closed 2023-05-12 by deyouslamtec