Slamtec/rplidar_ros #37: Question: How to launch RPLidar A3 in "outdoor" mode?

milan-r-shah, 2020-10-12:
The website (https://www.slamtec.com/en/Lidar/A3) says 

> RPLIDRA A3 supports to work under two modes alternatively: enhanced mode and outdoor mode.

However, when I checked the launch file **rplidar_a3.launch**, there is a parameter called `scan_mode`, and its `value` is set to `Sensitivity`. So, this value corresponds to which mode: **enhanced** mode or **outdoor** mode? 
Which string should I pass as a `value` to this `scan_mode` parameter to get **outdoor** mode?

eslamsherif5, 2020-11-03:
Hi, @milan-r-shah 
Have you found any info on this issue?
I was thinking of buying a RPLiDAR A3, but I'm now very hesitated after seeing many open issues and not-to-date ros driver.

milan-r-shah, 2020-11-07:
Hi @eslamsherif5 
Sorry for the late reply. I haven't found the exact solution but my colleague found the following code, upon digging the SDK code:
```
#   define RPLIDAR_CONF_SCAN_COMMAND_STD            0
#   define RPLIDAR_CONF_SCAN_COMMAND_EXPRESS        1
#   define RPLIDAR_CONF_SCAN_COMMAND_HQ             2
#   define RPLIDAR_CONF_SCAN_COMMAND_BOOST          3
#   define RPLIDAR_CONF_SCAN_COMMAND_STABILITY      4
#   define RPLIDAR_CONF_SCAN_COMMAND_SENSITIVITY    5
```
So, updated the `value` of `scan_mode` parameter in `rplidar_a3.launch` using above code:
https://github.com/Slamtec/rplidar_ros/blob/4f899e670bec2c9e1f26b0969f2de36d23618ef3/launch/rplidar_a3.launch#L8

Hope this helps!

billynugrahas, 2021-02-25:
Hi, have you found how to launch the outdoor mode?