Slamtec/rplidar_ros #40: Will there be a release to Noetic so it can be installed from binaries?

Timple, 2020-11-23:
Will there be a release to Noetic so it can be installed from binaries?

_Originally posted by @simonbogh in https://github.com/Slamtec/rplidar_ros/issues/33#issuecomment-714505658_

Since installing from apt-get is not the same as the question if the code works.

Timple, 2021-12-24:
Are there any objections if I do the release?

dimaqq, 2022-01-24:
poke

Timple, 2022-01-24:
I see that releases for other distributions also have been done by (it seems) third-party people.
I am not hearing any objections here, so I will go ahead and release for Noetic.

Timple, 2022-01-24:
https://github.com/ros/rosdistro/pull/31904

Timple, 2022-01-29:
Available in the testing repositories. Will be available in the next Noetic sync.

Please go ahead and test the binaries!

closed 2022-01-29 by Timple