Slamtec/rplidar_ros #41: Lidar Laserscan does not match with real measurements

jcgarciaca, 2020-12-04:
I am using two RPLidar A3 and one A2 for three different robots, but the laserscan between the A3 devices does not match (the laserscans from A2 and one of the A3 match well). I checked that both robots with A3 lidars have same software versions, it is:
```
SDK version: 1.12.0
Firmware Ver: 1.27
Hardware Rev: 6
RPLidar health status: 0
current scan mode: Sensitivity, max_distance: 25.0 m, Point number: 15.9 K, angle compensate: 4
```
Is there anything wrong with this lidar, or is it required an extra configuration?

[image]

jash101, 2021-04-16:
hey @jcgarciaca were you able to solve this issue?

jcgarciaca, 2021-04-19:
hi @jash101, No. I tried to configure the lidar with its calibration software, but it didn't work

jash101, 2021-04-19:
Oh, maybe check if lidar's orientation has no pitch or roll, which could change the measurements