Slamtec/rplidar_ros #63: *** buffer overflow detected *** error. Any one knows what's going on? Thanks.

bytemani, 2021-11-08:
roslaunch rplidar_ros rplidar.launch
... logging to /home/jetson/.ros/log/6d81d10c-4069-11ec-83a4-401c8380b7e0/roslaunch-jetson-desktop-8743.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.43.242:34569/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.12
 * /rplidarNode/angle_compensate: True
 * /rplidarNode/frame_id: laser
 * /rplidarNode/inverted: False
 * /rplidarNode/serial_baudrate: 115200
 * /rplidarNode/serial_port: /dev/ttyUSB0

NODES
  /
    rplidarNode (rplidar_ros/rplidarNode)

auto-starting new master
process[master]: started with pid [8834]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 6d81d10c-4069-11ec-83a4-401c8380b7e0
process[rosout-1]: started with pid [8856]
started core service [/rosout]
process[rplidarNode-2]: started with pid [8865]
[ INFO] [1636358212.891720298]: RPLIDAR running on ROS package rplidar_ros, SDK Version:2.0.0
*** buffer overflow detected ***: /home/jetson/rplidar_ws/devel/lib/rplidar_ros/rplidarNode terminated
[rplidarNode-2] process has died [pid 8865, exit code -6, cmd /home/jetson/rplidar_ws/devel/lib/rplidar_ros/rplidarNode __name:=rplidarNode __log:=/home/jetson/.ros/log/6d81d10c-4069-11ec-83a4-401c8380b7e0/rplidarNode-2.log].
log file: /home/jetson/.ros/log/6d81d10c-4069-11ec-83a4-401c8380b7e0/rplidarNode-2*.log

shamengjun, 2021-11-10:
sudo chmod 777 /dev/ttyUSB0 就是串口没有权限，串口文件根据你自己的修改

scalaros, 2023-01-23:
I was struggling with the same issue with rp_lidar and in my case it was usb cable to the usb-uart converter. I checked something like 12 different usb cables and two of them worked fine.
try:
ls -l /dev | grep ttyUSB
if there's no response then try another USB cable