Slamtec/rplidar_ros #67: Problems with SDK v2.0 and rplidar A2

ivrolan, 2021-11-26:
I was testing the rplidar A2 in noetic, Ubuntu 20.04. The messages LaserScan have range_max = 0. Rviz doesn't show the messages it receives. Although I can receive data, I think those points are not reliable.

Output of rplidar.launch or view_rpilidar.launch
`[ INFO] [1637658089.592109579]: RPLIDAR running on ROS package rplidar_ros, SDK Version:2.0.0
[ INFO] [1637658089.652513069]: RPLIDAR S/N: BED2B4C1E9839EF2C0E69EF767651F05
[ INFO] [1637658089.652736977]: Firmware Ver: 1.20
[ INFO] [1637658089.652854479]: Hardware Rev: 2
[ INFO] [1637658089.704478956]: RPLidar health status : OK.
[ INFO] [1637658092.661022504]: current scan mode: , sample rate: 2147483647 Khz, max_distance: 0.0 m, scan frequency:10.0 Hz`

Going back to a previous state of the repo, with SDK v1.9 solved this.

`[ INFO] [1637924003.395894798]: RPLIDAR running on ROS package rplidar_ros. SDK Version:1.9.0
RPLIDAR S/N: BED2B4C1E9839EF2C0E69EF767651F05
[ INFO] [1637924003.902991003]: Firmware Ver: 1.20
[ INFO] [1637924003.903030646]: Hardware Rev: 2
[ INFO] [1637924003.904721558]: RPLidar health status : 0
*WARN* YOU ARE USING DEPRECATED API: checkExpressScanSupported(bool&,_u32), PLEASE MOVE TO getAllSupportedScanModes()
*WARN* YOU ARE USING DEPRECATED API: getSampleDuration_uS, PLEASE MOVE TO RplidarScanMode::us_per_sample
[ INFO] [1637924004.429646224]: current scan mode: Express, max_distance: 16.0 m, Point number: 4.0K , angle_compensate: 1`

JinXiangLai, 2021-12-09:
Get it! Thanks. What a Hidden bug!!!

BobLeenb, 2022-11-25:
I got the same problem with rplidar A2 in noetic,Ubuntu20.04. And I choose to switch to tag 1.7.0 in rplidar_ros repo, the problem has be solved.

SquirminatorHotHead, 2023-03-08:
> I got the same problem with rplidar A2 in noetic,Ubuntu20.04. And I choose to switch to tag 1.7.0 in rplidar_ros repo, the problem has be solved.

Yes. This can fix the problem. It seems like a bug with the current version.

Alfarrow, 2023-05-16:
Thaaaanks I had this problem with A1 and this fixed it :)