Slamtec/rplidar_ros #71: node.cpp   scan_msg.range_max  is 0 with lidar A1

blusser31, 2021-12-19:
I could not get catorgapher to run, because publish_scan() sets  scan_msg.range_max to zero.
Output of `roslaunch rplidar_ros view_rplidar.launch` of line 315

` ROS_INFO("current scan mode: %s, sample rate: %d Khz, max_distance: %.1f m, scan frequency:%.1f Hz, ", current_scan_mode.scan_mode,(int)(1000/current_scan_mode.us_per_sample+0.5),max_distance, scan_frequency); `

max_distance is always zero.

os:
`my@my-VirtualBox:~$ uname -a
Linux my-VirtualBox 5.11.0-43-generic #47~20.04.2-Ubuntu SMP Mon Dec 13 11:06:56 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
`
Lidar A1 was included with:
`sudo chmod 777 /dev/ttyUSB0`

phithanhdanh, 2022-05-25:
I have this issue too

blusser31, 2022-05-25:
Since this issue is half a year old I don't exactly remember how to fix it.
But as I remember you have to go into line 79 in this file
https://github.com/Slamtec/rplidar_ros/blob/master/src/node.cpp
and replace `max_distance` with a number greater than 0.
So just put in the 8 and it works.
After that you have to recompile your catkin_ws however you compiled it.

cheers,
blusser