Slamtec/rplidar_ros #73: How to read the S2 under ROS2?

RemiFabre, 2022-01-18:
Hi,
The branch "ROS2" doens't seem to have support for the S2.
The branch "release/v2.0.0" seems to have support for the S2 under ROS1.

What is the recommended way to use the S2 under ROS2?
Best,

RemiFabre, 2022-01-19:
Update : 
Went on the branch "ROS2" and added a launch file inspired from "view_rplidar_s1_launch.py" but with a baudrate of 1000000:
```

def generate_launch_description():
    serial_port = LaunchConfiguration('serial_port', default='/dev/ttyUSB0')
    serial_baudrate = LaunchConfiguration('serial_baudrate', default='1000000') 
    frame_id = LaunchConfiguration('frame_id', default='laser')
    inverted = LaunchConfiguration('inverted', default='false')
    angle_compensate = LaunchConfiguration('angle_compensate', default='true')
```

The script is able to communicate with the LIDAR since I get it's S/N and firmware version, but it fails at starting the motor : 

> [rplidar_scan_publisher-1] [INFO] [1642585917.964269822] [rplidar_scan_publisher]: RPLidar health status : 0
> [rplidar_scan_publisher-1] [ERROR] [1642585919.968389524] [rplidar_scan_publisher]: Can not start scan: 80008000!

Any inputs on this?

RemiFabre, 2022-01-20:
Update, this partially fixed the issue:
Addind a "sleep(1);" in file "rplidar_scan_publisher.cpp", right after the line 301 (after the call to "drv->startMotor();").

I can now read the S2 under ROS2, but the number of points per scan is incorrect (most of the points the lidar records are not in the ROS message) and the max distance is incorrect (too low).

Chandan-kumar-R, 2023-04-03:
> Update, this partially fixed the issue: Addind a "sleep(1);" in file "rplidar_scan_publisher.cpp", right after the line 301 (after the call to "drv->startMotor();").
>
> I can now read the S2 under [...]

Hi,
 did you figured it out how to get the correct max_distance for S2
any input on this?

RemiFabre, 2023-04-10:
Try this:
https://github.com/Slamtec/sllidar_ros2

closed 2023-06-12 by deyouslamtec