Slamtec/rplidar_ros #77: ROS2 Dashing error

CharlesXu1124, 2022-02-23:
When I tried to launch rplidar_launch.py after building, it gives me the following error:
[rplidar_scan_publisher-1] [ERROR] [rplidar_scan_publisher]: Error, operation time out. RESULT_OPERATION_TIMEOUT! 
[rplidar_scan_publisher-1] [INFO] [rplidar_scan_publisher]: RPLIDAR running on ROS2 package rplidar_ros2. ROS2 SDK Version:1.0.1, RPLIDAR SDK Version:1.7.0
[ERROR] [rplidar_scan_publisher-1]: process has died [pid 5359, exit code 255, cmd '/home/charles/ros2_ws/install/rplidar_ros2/lib/rplidar_ros2/rplidar_scan_publisher __params:=/tmp/launch_params_nyh87764'].

Anyone knowing how to fix this?

deyouslamtec, 2023-05-18:
Check serial port permissions and baud rate used

CharlesXu1124, 2023-05-18:
nvm, resolved long time ago

closed 2023-05-18 by deyouslamtec