Slamtec/rplidar_ros #78: can't get S2 working with ros

Mdbirley, 2022-03-12:
I downloaded the latest package from the repo

It works great with an A1 however is fails with S2

Can you please give me an idea of what to do next

maxwell@maxwell-desktop:~/catkin_ws$ roslaunch rplidar_ros rplidar_s2.launch
... logging to /home/maxwell/.ros/log/42af7b2a-a223-11ec-9b4f-b827eb4c5002/roslaunch-maxwell-desktop-6484.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://maxwell-desktop:38934/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.17
 * /rplidarNode/angle_compensate: True
 * /rplidarNode/frame_id: laser
 * /rplidarNode/inverted: False
 * /rplidarNode/serial_baudrate: 1000000
 * /rplidarNode/serial_port: /dev/ttyUSB0

NODES
  /
    rplidarNode (rplidar_ros/rplidarNode)

auto-starting new master
process[master]: started with pid [6494]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 42af7b2a-a223-11ec-9b4f-b827eb4c5002
process[rosout-1]: started with pid [6507]
started core service [/rosout]
process[rplidarNode-2]: started with pid [6511]
[ INFO] [1647103290.834698241]: RPLIDAR running on ROS package rplidar_ros, SDK Version:2.0.0
[ INFO] [1647103291.395771696]: RPLIDAR S/N: B542ED93C5E19FD9B5E192F935904B72
[ INFO] [1647103291.395916477]: Firmware Ver: 1.00
[ INFO] [1647103291.395982357]: Hardware Rev: 18
[ERROR] [1647103291.447418643]: Error, rplidar internal error detected. Please reboot the device to retry.
[rplidarNode-2] process has died [pid 6511, exit code 255, cmd /home/maxwell/catkin_ws/devel/lib/rplidar_ros/rplidarNode __name:=rplidarNode __log:=/home/maxwell/.ros/log/42af7b2a-a223-11ec-9b4f-b827eb4c5002/rplidarNode-2.log].
log file: /home/maxwell/.ros/log/42af7b2a-a223-11ec-9b4f-b827eb4c5002/rplidarNode-2*.log

vivek2019, 2022-05-25:
Hey @Mdbirley 
I faced the similar error while working on S2, Based on the error, its the health of the Lidar and it got resolved when i cleaned the Lidar. You can try if that helps

vivek2019, 2022-05-26:
@Mdbirley We are facing the same issue, But also sometimes the lidar works as well, We are not able to find any specific combination that works reliable.

Did you find any solution for this issue ?

blusser31, 2022-07-03:
You need to have a baudrate of 256000 set on the hardware switch.
But this is the case with us as well.
We could not figure out a specific thing.

aghasaadmohammad, 2023-08-28:
I am using arduino mega to run s2 rplidar but unable to do that any success or any guidline?

3bdul1ah, 2024-09-23:
I'm currently facing an issue with my RPLidar S2 on the Jetson Nano. The lidar works perfectly fine on my laptop, but when I use it with the Jetson Nano, I consistently get the following error:

Error: rplidar internal error detected. Please reboot the device to retry.

I've tried the following solutions:

- Granted proper permissions (chmod for /dev/ttyUSB0)
- Ensured proper wiring, including separate power supply
- Rebooting the device and reconnecting multiple times
- Running sudo apt update && sudo apt upgrade
- Plugging and unplugging the device
When I unplug and reconnect, it works for about 2 seconds and then fails with the same internal error.

This seems to be a common problem with the S2 lidar, as many users have reported similar issues. Has anyone managed to figure out the exact reason for this behavior or found a reliable solution?