Slamtec/rplidar_ros #83: Service stop_motor is not working with 2.0.0

MichaelT64, 2022-04-13:
Hi,
The service "stop_motor" is not working with release 2.0.0 for Noetic and at least A1. There is a message in the debug-log after calling the service and also the expected empty response but the motor is spinning further.
Going back to version 1.7.0 it works again.

Michael

Lukewal99, 2023-07-09:
Just to agree, /stop motor isn't working in 2.0.0 :)  

```
pi@pi-ros-luke:~/edukit_bot_ws/src/rplidar_ros/src$ rosservice call /stop_motor "{}"
[DEBUG] [1688888391.894393248]: Stop motor

pi@pi-ros-luke:~/edukit_bot_ws/src/rplidar_ros/src$
```

Thanks very much

MaximBolduc, 2023-10-12:
Hi, we had the same problem with the latest ROS1 version. We simply forked and added the stop command like they did in the ROS2 version and now it works. We're not doing a PR since PR seem stale for ROS1...

[image]