Slamtec/rplidar_ros #84: how to increase the rplidar S2 publishing frequency upto 15Hz?

debanik123, 2022-05-07:
how to increase the rplidar S2 publishing frequency upto 15Hz?
Rplidar current publish frequency is only 10Hz. However the publishing frequency of rplidar A2 is 13 to 15 Hz.

kbln, 2023-04-30:
I have the same question! anyone can help please?

Nikhil619, 2023-12-05:
Were you able to solve it? 
If so,
Did you change the frequency in the launch file or....
What exactly did you do?

Thanks in advance

deyouslamtec, 2024-01-08:
Modify the scan_frequency parameter in the corresponding launch file.
[image]

kosmonauta144, 2024-01-08:
@deyouslamtec  I am struggling with changing the frequency on ros2 humble and I have included `rplidar_s1_launch.py` this way
```
    rplidar_launch = IncludeLaunchDescription(
            PythonLaunchDescriptionSource([str(rplidar_launch_src), '/rplidar_s1_launch.py']),
                launch_arguments={'frame_id': 'laser_link', 'serial_port': '/dev/laser_usb', 'scan_frequency':'15.0'}.items(),
        )
```
`frame_id` and `serial_port` parameters are working, but `scan_frequency` is not, because after launching I received message like these

```
[rplidar_node]: current scan mode: DenseBoost, sample rate: 9 Khz, max_distance: 40.0 m, scan frequency:10.0 Hz,
```
also frequency of the `/scan` topic is closely about 10 Hz

changing with launch parameter is not working in my case and ros2 launch files.

darkhannibal, 2024-01-08:
I am working with rplidar s1 on ROS1 noetic and after updating launch file in the way @deyouslamtec suggested but still the lidar is not spinning any faster.

While launching there is an info: `set lidar scan frequency to 10.0 Hz(600.0 Rpm)`
I've also checked `rostopic hz /scan` but it gives me reading of about 10 Hz

deyouslamtec, 2024-01-10:
I tested under Ubuntu20.04 ROS noetic. rplidar S1, S2, and S3 can adjust the publishing frequency by modifying scan_frequenc in the launch file.
[image]
[image]

darkhannibal, 2024-01-10:
This is content of modified launch I am using:
```
<launch>
  <node name="rplidarNode"          pkg="rplidar_ros"  type="rplidarNode" output="screen">
    <param name="serial_port"         type="string" value="/dev/ttyUSB0"/>  
    <param name="serial_baudrate"     type="int"    value="256000"/>
    <param name="frame_id"            type="string" value="laser"/>
    <param name="inverted"            type="bool"   value="false"/>
    <param name="angle_compensate"    type="bool"   value="true"/>
    <param name="scan_frequncy"	     type="double" value="15.0"/>
  </node>
</launch>
```
after launching it I get
[image]
So the ros parameter is being set, but does not affect the lidar

deyouslamtec, 2024-01-11:
@darkhannibal 'scan_frequency' not 'scan_frequncy'

darkhannibal, 2024-01-11:
@deyouslamtec Thanks man, now it works like a charm. I have not noticed that typo :/

kosmonauta144, 2024-01-11:
@deyouslamtec  does the same work on ROS2? Cause I tried to add this parameter to launch with default value of 15.0 and it does not change anything :crying_cat_face: 

```
def generate_launch_description():
    channel_type = LaunchConfiguration('channel_type', default='serial')
    serial_port = LaunchConfiguration('serial_port', default='/dev/ttyUSB0')
    serial_baudrate = LaunchConfiguration('serial_baudrate', default='256000') 
    frame_id = LaunchConfiguration('frame_id', default='laser')
    inverted = LaunchConfiguration('inverted', default='false')
    angle_compensate = LaunchConfiguration('angle_compensate', default='true')
    scan_frequency = LaunchConfiguration('scan_frequency', default='15.0')

    return LaunchDescription([
        DeclareLaunchArgument(
            'channel_type',
            default_value=channel_type,
            description='Specifying channel type of lidar'),
        
        DeclareLaunchArgument(
            'serial_port',
            default_value=serial_port,
            description='Specifying usb port to connected lidar'),

        DeclareLaunchArgument(
            'serial_baudrate',
            default_value=serial_baudrate,
            description='Specifying usb port baudrate to connected lidar'),
        
        DeclareLaunchArgument(
            'frame_id',
            default_value=frame_id,
            description='Specifying frame_id of lidar'),

        DeclareLaunchArgument(
            'inverted',
            default_value=inverted,
            description='Specifying whether or not to invert scan data'),

        DeclareLaunchArgument(
            'angle_compensate',
            default_value=angle_compensate,
            description='Specifying whether or not to enable angle_compensate of scan data'),

        DeclareLaunchArgument(
            'scan_frequency',
            default_value=scan_frequency,
            description='Specifying whether or not to enable angle_compensate of scan data'),

        Node(
            package='rplidar_ros',
            executable='rplidar_node',
            name='rplidar_node',
            parameters=[{'channel_type':channel_type,
                         'serial_port': serial_port,
                         'serial_baudrate': serial_baudrate,
                         'frame_id': frame_id,
                         'inverted': inverted,
                         'angle_compensate': angle_compensate,
                         'scan_frequency': scan_frequency}],
            output='screen'),
    ])
   ```
   
[image]

Parameters also do not work to increase the frequency

[image]

deyouslamtec, 2024-01-12:
@kosmonauta144 It doesn't work properly on ROS2 yet, wait for me to adjust the code

deyouslamtec, 2024-01-24:
@kosmonauta144 scan_frequcund can work on ROS2, please update the ros2 branch

kosmonauta144, 2024-01-24:
thank you very much, it works with parameter from launch file :smile:

closed 2024-02-18 by deyouslamtec