Slamtec/rplidar_ros #86: Can not start scan: 80008002!

M8ias, 2022-05-22:
I am currently using a RPLidar A1 with a Raspberry Pi 4B and keep getting this error when i launch the ROS-node 

`
[ INFO] [1653237839.801351212]: RPLIDAR running on ROS package rplidar_ros, SDK Version:2.0.0 `
`[ INFO] [1653237839.850478884]: RPLIDAR S/N: E08CE7F2C1E29BF3C0E39EF6115F4936 `
`[ INFO] [1653237839.850556255]: Firmware Ver: 1.15 `
`[ INFO] [1653237839.850585624]: Hardware Rev: `
`[ INFO] [1653237839.901918818]: RPLidar health status : OK.`
`[ERROR] [1653237844.842084127]: Can not start scan: 80008002!`

Some googling makes it seem like a power-issue, but after testing with three different usb-cables on the pi, as well as two separate computers the issue still persists. I guess all the usb cables may be awful, but it also seems like the ones with power issues got the error code `[ERROR] [1653237844.842084127]: Can not start scan: 80008000!``. Any tips on what the code means?

vivek2019, 2022-05-25:
This code is generally for the Power requiremts. Is the lidar properly powered?  There are 2 cables connected to the lidar, one for Data and one for power, if not connected try connecting it, Might get resolved

jcocovich, 2022-11-12:
Actually, the model mentioned, I have the same one, is powered and communicates through an external converter card that is supplied with the scanner. Plugging this USB card in to the RPI4 USB port for power and communication. My RPI4 is powered through the GPIO pins from an onboard controller.
I get this error all the time unless I power the RPI4 through the USB-c power connector. Seems to me to be a power issue but NOTHING works but the power supply. Doesn't make for a very portable device unless you have an extremely long extension cord.
I don't believe this is how the RPLidar is supposed to work but I haven't found any intelligent responses anywhere yet. Good Luck. If my RPLidar reports a health status of OK, I EXPECT it to scan. Timing or noise is a possible rat hole I could pursue.

minhaj6, 2023-04-13:
i get the same error, using virtualbox on a recent huawei laptop.

firstbot1, 2023-10-06:
I modified the code to force a scan and the problem was resolved.
sl_lidar_driver.cpp(361line)
return startScanExpress(true, typicalMode, 0, outUsedScanMode);

knnurl, 2023-12-27:
My rplidar a1m1 was working very well with me MSI laptop and I was able to see the laser scan in rviz. When I plugged it on RPi4, I had the following error:

   theta: 359.99 Dist: 16383.75 
S  theta: 359.99 Dist: 1073741824.00 
Error, cannot grab scan data.

Some people recommended changing connection cables but it only made the case worse for me, if not better. With another cable and rplidar sdk command: 

./simple_grabber --channel --serial /dev/ttyUSB0 115200, 
I had the following error:

Error, unexpected error, code: 80008004
Segmentation fault (core dumped)

I'm about to give up on this until I get back to laptop.

deyouslamtec, 2024-01-09:
It is recommended to use a high-power power adapter to power Raspberry Pi 4

COSIME56, 2024-02-10:
#84

423Tech, 2025-10-24:
Is there any solution to this error code? I'm using a CH340 chip to communicate with A3M1-R4 Lidar; whether or not I connect the MOTORCTL, the problem still happens. (I tried both 3.3V and 5V directly connected to MOTORCTL. If this is the reason, pls tell me too.)

env: Ubuntu 24.02, RDK X5, SLLIDAR SDK Version:2.1.0

Here's slider_ros2's output:
`ros2 launch sllidar_ros2 sllidar_a3_launch.py
[INFO] [launch]: All log files can be found below /home/sunrise/.ros/log/2025-10-24-21-59-52-260094-ubuntu-12243
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [sllidar_node-1]: process started with pid [12261]
[sllidar_node-1] [INFO] [1761314392.615664522] [sllidar_node]: SLLidar running on ROS2 package SLLidar.ROS2 SDK Version:1.0.1, SLLIDAR SDK Version:2.1.0
[sllidar_node-1] [INFO] [1761314393.124504031] [sllidar_node]: SLLidar S/N: ABCDEFGHIJKLMN (PS: fake code)
[sllidar_node-1] [INFO] [1761314393.124878193] [sllidar_node]: Firmware Ver: 1.32
[sllidar_node-1] [INFO] [1761314393.125058858] [sllidar_node]: Hardware Rev: 6
[sllidar_node-1] [INFO] [1761314393.128050362] [sllidar_node]: SLLidar health status : 0
[sllidar_node-1] [INFO] [1761314393.128236318] [sllidar_node]: SLLidar health status : OK.
[sllidar_node-1] [ERROR] [1761314395.206243263] [sllidar_node]: Can not start scan: 80008002!
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[sllidar_node-1] [INFO] [1761314409.313532235] [sllidar_node]: Stop motor
[INFO] [sllidar_node-1]: process has finished cleanly [pid 12261]`

Here's USB output:
`ls -l /dev/ttyUSB*
dmesg | tail -n 40
crw-rw---- 1 root dialout 188, 0 Oct 24 22:00 /dev/ttyUSB0
[ 1790.926978] usb 1-1.2: New USB device found, idVendor=1a86, idProduct=7522, bcdDevice= 2.64
[ 1790.927036] usb 1-1.2: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 1790.927068] usb 1-1.2: Product: USB Serial
[ 1790.987384] ch341 1-1.2:1.0: ch341-uart converter detected
[ 1790.988302] usb 1-1.2: ch341-uart converter now attached to ttyUSB0
[ 1859.908771] usb 1-1.2: USB disconnect, device number 4
[ 1859.910192] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[ 1859.910386] ch341 1-1.2:1.0: device disconnected
[ 1872.775664] usb 1-1.2: new full-speed USB device number 5 using xhci-hcd
[ 1872.926755] usb 1-1.2: New USB device found, idVendor=1a86, idProduct=7522, bcdDevice= 2.64
[ 1872.926810] usb 1-1.2: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 1872.926843] usb 1-1.2: Product: USB Serial
[ 1872.994717] usb 1-1.2: can't set config #1, error -71
[ 1873.000642] usb 1-1.2: USB disconnect, device number 5
[ 1890.565639] usb 1-1.2: new full-speed USB device number 7 using xhci-hcd
[ 1890.716759] usb 1-1.2: New USB device found, idVendor=1a86, idProduct=7522, bcdDevice= 2.64
[ 1890.716814] usb 1-1.2: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 1890.716847] usb 1-1.2: Product: USB Serial
[ 1890.753179] ch341 1-1.2:1.0: ch341-uart converter detected
[ 1890.753837] usb 1-1.2: ch341-uart converter now attached to ttyUSB0
[ 1899.840687] usb 1-1.2: USB disconnect, device number 7
[ 1899.842051] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[ 1899.842242] ch341 1-1.2:1.0: device disconnected
[ 1907.635617] usb 1-1.2: new full-speed USB device number 9 using xhci-hcd
[ 1907.755704] usb 1-1.2: Device not responding to setup address.
[ 1907.975650] usb 1-1.2: Device not responding to setup address.
[ 1908.195731] usb 1-1.2: device not accepting address 9, error -71
[ 1909.065622] usb 1-1.2: new full-speed USB device number 10 using xhci-hcd
[ 1909.216763] usb 1-1.2: New USB device found, idVendor=1a86, idProduct=7522, bcdDevice= 2.64
[ 1909.216815] usb 1-1.2: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 1909.216848] usb 1-1.2: Product: USB Serial
[ 1909.311120] usb 1-1.2: can't set config #1, error -32
[ 1909.318735] usb 1-1.2: USB disconnect, device number 10
[ 1909.795527] usb 1-1.2: new full-speed USB device number 11 using xhci-hcd
[ 1949.985602] usb 1-1.2: new full-speed USB device number 12 using xhci-hcd
[ 1950.136658] usb 1-1.2: New USB device found, idVendor=1a86, idProduct=7522, bcdDevice= 2.64
[ 1950.136711] usb 1-1.2: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 1950.136745] usb 1-1.2: Product: USB Serial
[ 1950.171178] ch341 1-1.2:1.0: ch341-uart converter detected
[ 1950.172472] usb 1-1.2: ch341-uart converter now attached to ttyUSB0`