Slamtec/rplidar_ros #87: Error, rplidar internal error detected. Please reboot the device to retry.

vivek2019, 2022-05-25:
Hello

I am using the new Rplidar s2 on jetson nx, I frequently face this below issue, what should i do to overcome this 

roslaunch rplidar_ros rplidar_s2.launch 
... logging to /home/nvidia/.ros/log/e8398694-dc20-11ec-a98c-00181af28bd5/roslaunch-tegra-ubuntu-19279.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.104:36973/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /rplidarNode/angle_compensate: True
 * /rplidarNode/frame_id: laser
 * /rplidarNode/inverted: False
 * /rplidarNode/serial_baudrate: 1000000
 * /rplidarNode/serial_port: /dev/rplidar

NODES
  /
    rplidarNode (rplidar_ros/rplidarNode)

auto-starting new master
process[master]: started with pid [19345]
ROS_MASTER_URI=http://192.168.1.104:11311

setting /run_id to e8398694-dc20-11ec-a98c-00181af28bd5
process[rosout-1]: started with pid [19374]
started core service [/rosout]
process[rplidarNode-2]: started with pid [19382]
[ INFO] [1653479445.408398218]: RPLIDAR running on ROS package rplidar_ros, SDK Version:2.0.0
[ INFO] [1653479445.464188214]: RPLIDAR S/N: 5AACECF8C4E699D5B8EB99F9D6734715
[ INFO] [1653479445.464808939]: Firmware Ver: 1.01
[ INFO] [1653479445.464974537]: Hardware Rev: 18
[ERROR] [1653479445.517251953]: Error, rplidar internal error detected. Please reboot the device to retry.
[rplidarNode-2] process has died [pid 19382, exit code 255, cmd /home/nvidia/aurus_ws/install/rplidar_ros/lib/rplidar_ros/rplidarNode __name:=rplidarNode __log:=/home/nvidia/.ros/log/e8398694-dc20-11ec-a98c-00181af28bd5/rplidarNode-2.log].
log file: /home/nvidia/.ros/log/e8398694-dc20-11ec-a98c-00181af28bd5/rplidarNode-2*.log
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Sudharsan10, 2022-07-06:
I'm facing the same error and only difference is that I used a raspberry Pi 4B

jcrupi, 2022-07-14:
Any resolutions?

vivek2019, 2022-07-15:
The system worked for me.

I connected the same device with different s2 adapter and cabel and it
worked.

After that i used the same old wires and cables and it worked fine. I
haven't faced issue after that.

astumpf, 2022-08-22:
I'm experiencing the same error message with 2 brand-new devices. Cannot get them to work at all. I appreciate any ideas how to solve this issue.

vivek2019, 2022-08-22:
Hey

I bought a new Lidar and fortunately, both the S2 lidar worked after
connecting the old lidar with the new cable. I also tried multiple
configurations with cables, adapters and lidar and all the configurations
worked. So I am not sure what was the reason but now both Lidars are
working fine.

CycleMark, 2023-05-19:
Has this been solved at all. I get this issue on any board I try. My A3 and A1 SLAMTec sensors work just fine. I can't be having a sensor which may or may not work on a whim (well tbf it mostly doesn't work)

Thanks

Mark

j953302, 2023-05-23:
Hello, I use S2 and face the same problem randomly. Sometimes it work fine but sometimes it doesn't. I have give it independent 5V power but it doesn't figure it out. Is there any other advice? Thank.

SyJarvis, 2023-11-27:
Try providing additional power supply

CycleMark, 2023-11-27:
Unfortunately external PSU makes no difference. Having to remove and reboot the sensor makes is somewhat useless

SyJarvis, 2023-11-27:
21软件7班-你的邮箱已收到

3bdul1ah, 2024-09-23:
I'm currently facing an issue with my RPLidar S2 on the Jetson Nano. The lidar works perfectly fine on my laptop, but when I use it with the Jetson Nano, I consistently get the following error:

Error: rplidar internal error detected. Please reboot the device to retry.

I've tried the following solutions:

- Granted proper permissions (chmod for /dev/ttyUSB0)
- Ensured proper wiring, including separate power supply
- Rebooting the device and reconnecting multiple times
- Running sudo apt update && sudo apt upgrade
- Plugging and unplugging the device
When I unplug and reconnect, it works for about 2 seconds and then fails with the same internal error.

This seems to be a common problem with the S2 lidar, as many users have reported similar issues. Has anyone managed to figure out the exact reason for this behavior or found a reliable solution?

SyJarvis, 2024-09-23:
24计科(专升本)2班-你的邮箱已收到

CycleMark, 2024-10-12:
@3bdul1ah - sadly not. My sensor has spent 2 years in a box. Not heard anything from SLAMTEC. Like you I've tried everything

SyJarvis, 2024-10-12:
24计科(专升本)2班-你的邮箱已收到

3bdul1ah, 2024-10-12:
> @3bdul1ah - sadly not. My sensor has spent 2 years in a box. Not heard anything from SLAMTEC. Like you I've tried everything 

@CycleMark - I managed to solve the issue. I was using a Jetson Nano with Ubuntu 20.04, which isn’t natively supported as the Jetson Nano is optimized for Ubuntu 18.04. This mismatch likely caused some limitations, such as the USB port not supplying enough voltage to power the LiDAR, which could be why we kept seeing the error.

After downgrading to Ubuntu 18.04 and testing it, surprisingly, everything worked without any issues. From my experience, this seems to be a reasonable explanation. However, the error is quite general and could occur for different reasons, making it hard to pinpoint the exact cause.

While it’s working fine on the Jetson Nano with Ubuntu 18.04, I did notice that the error could also stem from changes in the device path, like /dev/ttyUSB*, which keeps switching between 1 and 2. This doesn’t seem like an issue that can be permanently fixed by assigning it to a single port. So, even after downgrading (which worked for me), you might still encounter the error, but likely due to the port switching.