Slamtec/rplidar_ros #91: Ros2-Humble Support missing

aagrawa3-dev, 2022-07-20:
**Ros2-Humble** is released in May22, I see no news on this repo for its support.
Please comment about plans to support it..

lopenguin, 2022-08-18:
Here's a repo with a couple of changes that allow the code to be compiled/work with ROS 2 Humble: https://github.com/lopenguin/rplidar_ros2 (see the ros2 branch).

Note that I also made a change to the start/stop motor service, combining the two into one service (toggle_motor). This can be removed by commenting out anything with toggle motor and uncommenting the start/stop motor services. See the commit for details.

mrmara, 2023-01-17:
On ROS Humble I get the following error:

```
terminate called after throwing an instance of 'rclcpp::exceptions::UninitializedStaticallyTypedParameterException'
  what():  Statically typed parameter 'channel_type' must be initialized.
```

deyouslamtec, 2023-05-12:
It has been fixed

closed 2023-05-12 by deyouslamtec