Slamtec/rplidar_ros #93: sudo chmod 777 /dev/ttyUSB0 就是串口没有权限，串口文件根据你自己的修改

girbish2, 2022-09-15:
sudo chmod 777 /dev/ttyUSB0 就是串口没有权限，串口文件根据你自己的修改

_Originally posted by @shamengjun in https://github.com/Slamtec/rplidar_ros/issues/63#issuecomment-964900164_
That didnt solve my problem, any other solution? why does this error ocurred

xeno00, 2022-11-01:
@girbish2 any fix on this yet?

BraulioSanchezH, 2022-11-01:
@xeno00 hi james, im girbish on another account, it seem that the sdk on the rplidar is not meant to be used on ubuntu foccal 20.04 on a raspberri,wich was my case, but the rosnode is...i was trying to use, release and debug this code on qtcreator, i did  not use the sdk, i used this repository on my catkin file, and make the catkin build process... the overflow problem happened to me because something else was using ttyUSB0, or the conecction of the lidar was persistant... so try to see on /dev if something else is connected on this port..... also, if sudo is using the connection, give permission of chmod each time you connect the lidar with sudo chmod 666 ttyUSB0,but you uually dont need sudo to use the lidar... let me know if i can help you in any other way...

kadlu, 2022-11-02:
using the standard addresses like /dev/tty* is not a save way to connect the lidar in case of you have other components connected. It can be that the system gives the name to someone else. Better is to generate a rules file which creates a unique name for the device. There is an example here: http://wiki.ros.org/rplidar under 3. Device settings. Hope that helps.

closed 2023-06-19 by deyouslamtec