Slamtec/rplidar_ros #105: Model A2M12

rootes1725, 2023-03-16:
Just a quick note for anyone experiencing issues using this package with the A2M12 device as there is a lack of documentation online for this specific version. 

The launch files associated with the A2 product line will not work with the M12 version, you will likely run into an error similar to:

`[rplidarNode-1] process has died [pid 8575, exit code 255....`

I think this is simply due to the baudrate defaulting to 115200 for all A2 versions whilst the M12 version runs at 256000. The good news here is that the A2M12 will work just fine if you run the A3 launch file as this defaults to the correct baudrate for this model.

jayro-zaid, 2023-04-27:
Hello! 

This worked for me, but only once! It doesn't work now, did you have a similar experience? I am using a virtual machine with ROS Noetic.

Thank you for your kind reply in advance!

deyouslamtec, 2023-06-12:
Thank you very much. Different launch files have been created for different models of lidar，and now they are clearer to use

closed 2023-06-12 by deyouslamtec