Slamtec/rplidar_ros #109: RPlidar S2E "Error, operation time out. RESULT_OPERATION_TIMEOUT!"  error

hsynklceng, 2023-04-06:
Hi,
I am trying to use rplidar s2e with Ubuntu 18.04 on Nvidia AGX. When i connect rplidar to agx, did not get any ip on ifconfig screen. But, i assign ip as `sudo ifconfig eth0 192.168.0.10 netmask 255.255.255.0` . And then, i tried to run rplidar_s2e.launch and i got this error.

How can i fix this error?

hsynklceng, 2023-04-06:
Hi,
i fixed this issue, I had to fix the ip to 192.168.11.x. Now, i want to use 2 lidar but i have 1 ip(192.168.11.2). How can i find other rplidar's ip?

WubinXia, 2023-05-05:
Please configure the IP of the lidar, for example , configure the first lidar IP to 192.168.11.3,  second lidar IP to 192.168.11.4
For details about the configuration method, see  the page 14 of S2E Dev Kit User Manual (https://www.slamtec.com/en/Support#rplidar-s2)

deyouslamtec, 2023-05-12:
I can open it, try it again