Slamtec/rplidar_ros #110: Lidar A2M12 - Error, operation time out. RESULT_OPERATION_TIMEOUT!

jayro-zaid, 2023-04-30:
Dear community,

A few weeks ago, the company where I am working acquired a Lidar A2M12, and I have followed these tutorials, but nothing works! 

a) http://wiki.ros.org/rplidar
b) https://github.com/robopeak/rplidar_ros/wiki

I want to explain to you step by step my procedure, in such a way that you can give me a solution or suggestion about this huge problem that I have, since we are in the middle of a project.

I am using a virtual machine with Ubuntu 20.04 and ROS Noetic. I am powering the lidar with a mobile phone charger of: 5V, 2A.

**Step 1**
First, I cloned the RPLIDAR ROS package (https://github.com/slamtec/rplidar_ros) in my catkin's workspace src folder:
[image]

Output:
[image]

Then, I ran `catkin_make` to build rplidarNode and rplidarNodeClient:
[image]

Output:
[image]
[image]

**Step 2**
I set the permissions for the rplidar device. For that, I opened the original file `rplidar.rules`: 
[image]

Output:
[image]

And then, I have done the settings below (according to http://wiki.ros.org/rplidar): 
[image]

I saved the new `rplidar.rules` file, and finally I ran the script `./create_udev_rules.sh`:

Output:
[image]

Now, I have the `rplidar.rules` file in `/etc/udev/rules.d`:
[image]

**Step 3**
Once I have changed the USB port remap, I changed the launch file about the serial_port value. This is the original `rplidar.launch` file:
[image]
[image]

This is the new `rplidar.launch` file:
[image]

The same procedure with the `rplidar_a3.launch` file:
[image]

The original `rplidar_a3.launch` file:
[image]

The new `rplidar_a3.launch` file:
[image]

**Step 4**
I checked the authority of rplidar's serial-port, and this is the result:
[image]

Then, I added the authority to write:
[image]

**Step 5**
Finally, I started the rplidar node. `roscore` initialization without any problems.
[image]

**Error executing `view_rplidar.launch`!**
[image]

**I had read that in some cases the solution was to run `view_rplidar_a3.launch`, but I still get an error!**
[image]

Please, if someone could give me a suggestion, solution, or tell me about a similar experience, I would be very grateful!

I look forward to your comments. Thank you in advance!

GiladRobee, 2023-05-03:
i have the same issue in both ros2 foxy and ros2 humble (minus the ros1 specifics)

tony-slamtec, 2023-05-04:
It seems you are using RPLIDAR A2M12, whose baudrate should be 256000 instead of 115200. Please change serial_baudrate param in your launch file and try again.

And also, RPLIDAR consume much electricty, so please use a USB port that can provide more current.

tugbakara, 2023-05-30:
> 

Got the same but I couldn't solve with this package do you have any idea about it?

GiladRobee, 2023-05-31:
> Got the same but I couldn't solve with this package do you have any idea about it?
> 
> > **edit**, i found out its about the baud rate of the serial coms.
> > alternatively, you can use this: https://github.com/Slamtec/sllidar_ros2/tree/main
> > it works great for me

As mentioned before,  this issue is mainly related to baud-rate issues (didn't go deep enough to see exactly how, but it is so based on my very little experience). so check the baud-rates specification of your lidar and change the relevant parameters.
that should do the trick.

tugbakara, 2023-05-31:
My lidar is s2e so it uses LAN cable.

deyouslamtec, 2023-07-13:
> It seems you are using RPLIDAR A2M12, whose baudrate should be 256000 instead of 115200. Please change serial_baudrate param in your launch file and try again.
> 
> And also, RPLIDAR consume much electricty, so please use a USB port that can provide more current.

Just refer to this

closed 2023-07-13 by deyouslamtec