Slamtec/rplidar_ros #118: Buffer overflow detected for testing RPLIDAR A1M8 sensor

AdrianUIC, 2023-06-13:
Running into a Buffer Overflow detected warning when running the command "roslaunch rplidar_ros rplidar_a1.launch". I have tried to using solutions found to solve this issue but it still is happening despite it. If anyone can assist me on this, it would be greatly appreciated. 

I am on a windows PC using ROS melodic to test my RPLIDAR sensor using Ubuntu 18.04.06 LTS.

[image]

deyouslamtec, 2023-07-13:
I didn't reproduce it.How did you test it?
[image]

deyouslamtec, 2023-08-06:
I reproduced it。

closed 2023-08-06 by deyouslamtec