Slamtec/rplidar_ros #121: How to change default IP of Rplidar S2E

raviparmar11, 2023-06-23:
The S2E has a default IP address of 192.168.11.2 and a default port of 8089. This means that, in order to read data from the LIDAR using ROS on your local PC, your PC must also be on the same IP range as the LIDAR. This can be achieved by changing the IP address of your PC to be within the same IP range. For example, you can set your PC's IP address to 192.168.11.4.

So is there any way to change default ip address of lidar s2e.

deyouslamtec, 2023-06-25:
You can refer to page 14 of the [user manual of S2E](https://bucket-download.slamtec.com/5350555d005bfd7a8685fcea6e80e9edb764adb1/SLAMTEC_rplidarkit_usermanual_S2E_v1.0_en.pdf). Configure the IP address through the "Command > IP Config" option in the frame_grabber program.

raviparmar11, 2023-06-26:
Yes got it ..But I am using Jetson nx and the above SDK is for Winows. In manual they has also said that **"For SDK, see
setLidarIpConf and reset."** but I didn't find anything related setLidarIpConf for lunix SDK

deyouslamtec, 2023-06-26:
setLidarIpConf is declared in [sl_lidar_driver_impl.h](https://github.com/Slamtec/rplidar_sdk/blob/master/sdk/include/sl_lidar_driver_impl.h#L80) and defined in [sl_lidar_driver.cpp](https://github.com/Slamtec/rplidar_sdk/blob/master/sdk/src/sl_lidar_driver.cpp#L634).

raviparmar11, 2023-06-27:
Ok. But how to use it is there any manual for it.

deyouslamtec, 2023-06-27:
You can refer to the [SDK user manual](https://bucket-download.slamtec.com/6957283725b66750890024d1f0d12940fa079e06/LR002_SLAMTEC_rplidar_sdk_v2.0_en.pdf).
more information ：https://www.slamtec.com/en/Support#rplidar-s2

closed 2023-06-28 by deyouslamtec