Slamtec/rplidar_ros #128: Current Package Not In Rosdep

brianabouchard, 2023-07-25:
Hello! I've been sorting through some issues with mapping using the RPLidar A1M8. I noticed you recently made several key updates to this repository that should fix some of the issues I'm having. However, when I run a rosdep update, it says there is no update for this package. I did some digging and it looks like the current package registered with rosdep is [this fork](https://github.com/allenh1/rplidar_ros) of your repository. I raised an issue on that repo asking them to merge your recent commits and update rosdep, but I haven't heard anything back. Is there a way you can directly take over the rosdep package and update it with your new version? 

Side note: I know I can build from source in my workspace, but I'm trying to use this in an educational setting and it is much easier for students to be able to just install dependencies from rosdep. 

If I can be of any help in making this happen, please let me know! Thanks!

deyouslamtec, 2023-07-31:
Hi! rplidar_ros 2.1.3 for humble have been meraged([issue](https://github.com/ros/rosdistro/pull/38049#event-9904293811)).but  can only available in the ros-testing repository(https://repo.ros2.org/status_page/ros_humble_default.html?q=rplidar_ros).It will take about three weeks to see the official version.

brianabouchard, 2023-08-01:
Awesome, thanks so much!

closed 2023-08-01 by brianabouchard