Slamtec/rplidar_ros #132: catkin_make errors

r8drfn89, 2023-08-16:
I have done everything I can to install the ROS dependencies and now I am getting this error when trying to run catkin_make

~ /ros_catkin_ws/src $ catkin_make
Base path: /home/astro-scout/ros_catkin_ws/src
The specified source space "/home/astro-scout/ros_catkin_ws/src/src" does not exist

deyouslamtec, 2023-08-16:
please run `catkin_make ` in `~ /ros_catkin_ws`

r8drfn89, 2023-08-16:
[image]
This is the error after running catkin_make in ~ros_catkin_ws

r8drfn89, 2023-08-16:
[image]
I even have the empy installed

r8drfn89, 2023-08-16:
[image]

r8drfn89, 2023-08-16:
I tried this catkin_make -DPYTHON_EXECUTABLE=/usr/bin/pyhton3 -DPYTHON_INCLUDE_DIR=/usr/include/python3.7m and it gave me this message. 
[image]

So I tried doing what it recommended catkin_make_isolated -DPYTHON_EXECUTABLE=/usr/bin/pyhton3 -DPYTHON_INCLUDE_DIR=/usr/include/python3.7m   and it gives this error
[image]

deyouslamtec, 2023-08-17:
Try dev-sdk branch, the latest commit may solve this problem

r8drfn89, 2023-08-17:
@deyouslamtec do you understand what is going on?

deyouslamtec, 2023-08-18:
Can you tell me your device information and OS version？

r8drfn89, 2023-08-20:
I am currently using ubuntu 20.04 on the raspberry pi 4 armhf. I was able to get further in the catkin_make process than before but now I am facing other errors. Here is the output from the /home/ubuntu/ros_catkin_ws/build/CMakeFiles/CMakeError.log"

 Performing C SOURCE FILE Test CMAKE_HAVE_LIBC_PTHREAD failed with the following output:
Change Dir: /home/ubuntu/ros_catkin_ws/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_fc22e/fast && /usr/bin/make -f CMakeFiles/cmTC_fc22e.dir/build.make CMakeFiles/cmTC_fc22e.dir/build
make[1]: Entering directory '/home/ubuntu/ros_catkin_ws/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_fc22e.dir/src.c.o
/usr/bin/cc   -DCMAKE_HAVE_LIBC_PTHREAD   -o CMakeFiles/cmTC_fc22e.dir/src.c.o   -c /home/ubuntu/ros_catkin_ws/build/CMakeFiles/CMakeTmp/src.c
Linking C executable cmTC_fc22e
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_fc22e.dir/link.txt --verbose=1
/usr/bin/cc  -DCMAKE_HAVE_LIBC_PTHREAD    -rdynamic CMakeFiles/cmTC_fc22e.dir/src.c.o  -o cmTC_fc22e
/usr/bin/ld: CMakeFiles/cmTC_fc22e.dir/src.c.o: in function `main':
src.c:(.text+0x36): undefined reference to `pthread_create'
/usr/bin/ld: src.c:(.text+0x3e): undefined reference to `pthread_detach'
/usr/bin/ld: src.c:(.text+0x48): undefined reference to `pthread_join'
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_fc22e.dir/build.make:87: cmTC_fc22e] Error 1
make[1]: Leaving directory '/home/ubuntu/ros_catkin_ws/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_fc22e/fast] Error 2

Source file was:
#include <pthread.h>

void* test_func(void* data)
{
  return data;
}

int main(void)
{
  pthread_t thread;
  pthread_create(&thread, NULL, test_func, NULL);
  pthread_detach(thread);
  pthread_join(thread, NULL);
  pthread_atfork(NULL, NULL, NULL);
  pthread_exit(NULL);

  return 0;
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/ubuntu/ros_catkin_ws/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_2a1be/fast && /usr/bin/make -f CMakeFiles/cmTC_2a1be.dir/build.make CMakeFiles/cmTC_2a1be.dir/build
make[1]: Entering directory '/home/ubuntu/ros_catkin_ws/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_2a1be.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_2a1be.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.16/Modules/CheckFunctionExists.c
Linking C executable cmTC_2a1be
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_2a1be.dir/link.txt --verbose=1
/usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create    -rdynamic CMakeFiles/cmTC_2a1be.dir/CheckFunctionExists.c.o  -o cmTC_2a1be  -lpthreads
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_2a1be.dir/build.make:87: cmTC_2a1be] Error 1
make[1]: Leaving directory '/home/ubuntu/ros_catkin_ws/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_2a1be/fast] Error 2

Determining if the function trunc exists failed with the following output:
Change Dir: /home/ubuntu/ros_catkin_ws/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_0442e/fast && /usr/bin/make -f CMakeFiles/cmTC_0442e.dir/build.make CMakeFiles/cmTC_0442e.dir/build
make[1]: Entering directory '/home/ubuntu/ros_catkin_ws/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_0442e.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=trunc   -o CMakeFiles/cmTC_0442e.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.16/Modules/CheckFunctionExists.c
<command-line>: warning: conflicting types for built-in function ‘trunc’; expected ‘double(double)’ [-Wbuiltin-declaration-mismatch]
/usr/share/cmake-3.16/Modules/CheckFunctionExists.c:7:3: note: in expansion of macro ‘CHECK_FUNCTION_EXISTS’
    7 |   CHECK_FUNCTION_EXISTS(void);
      |   ^~~~~~~~~~~~~~~~~~~~~
/usr/share/cmake-3.16/Modules/CheckFunctionExists.c:1:1: note: ‘trunc’ is declared in header ‘<math.h>’
  +++ |+#include <math.h>
    1 | #ifdef CHECK_FUNCTION_EXISTS
Linking C executable cmTC_0442e
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_0442e.dir/link.txt --verbose=1
/usr/bin/cc  -DCHECK_FUNCTION_EXISTS=trunc    -rdynamic CMakeFiles/cmTC_0442e.dir/CheckFunctionExists.c.o  -o cmTC_0442e
/usr/bin/ld: CMakeFiles/cmTC_0442e.dir/CheckFunctionExists.c.o: in function `main':
CheckFunctionExists.c:(.text+0xa): undefined reference to `trunc'
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_0442e.dir/build.make:87: cmTC_0442e] Error 1
make[1]: Leaving directory '/home/ubuntu/ros_catkin_ws/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_0442e/fast] Error 2
[image]

[image]

r8drfn89, 2023-08-23:
@deyouslamtec I got it to work by getting a new sd card and installing the 64bit version of Ubuntu 20.04 onto my raspi 4. Then I used http://wiki.ros.org/noetic/Installation/Ubuntu again, along with your video. It helped a lot and it now works. Thanks for all of your help.

closed 2023-08-23 by r8drfn89