Slamtec/rplidar_ros #133: Error, cannot retrieve rplidar health code: 80008000

wsung1, 2023-08-17:
Hi, there,
I'm using S2 model based on RPLIDAR ROS package with ROS melodic and Ubuntu 18.04.
One unit works fine but multiple units throw an error:
[ERROR] [1692239374.076535278]: Error, cannot retrieve rplidar health code: 80008000
when launching with
  <include ns="lidar0" file="$(find rplidar_ros)/launch/rplidar_s2.launch">
    <param name="serial_port" value="/dev/ttyUSB0"/>
  </include>
  <include ns="lidar1" file="$(find rplidar_ros)/launch/rplidar_s2.launch">
    <param name="serial_port" value="/dev/ttyUSB1"/>
  </include>
As shown below, this error has been reported before but it was not sorted out properly.
How to connect two rplidars on one computer
https://answers.ros.org/question/207222/how-to-connect-two-rplidars-on-one-computer/
Do you have any idea?
Thanks for your interest.

YHuniv, 2023-12-22:
HI,

did you found the solution ? thanks

deyouslamtec, 2023-12-25:
You can use the following launch file. I tried it feasible
`
<launch>

    <group ns="rplidar1">
        <include  file="$(find rplidar_ros)/launch/rplidar_a2m12.launch" />
        <arg name="serial_port"  value="/dev/ttyUSB0"/>
    </group>
    <group ns="rplidar2">
        <include file="$(find rplidar_ros)/launch/rplidar_a2m8.launch" />
        <arg name="serial_port" value="/dev/ttyUSB1"/>
    </group> 

</launch>
`

deyouslamtec, 2023-12-25:
try it. I tried it feasible

<launch>

    <group ns="rplidar1">
        <include  file="$(find rplidar_ros)/launch/rplidar_a2m12.launch" />
        <param name="/rplidar1/rplidarNode/serial_port"  value="/dev/ttyUSB0"/>
    </group>
    <group ns="rplidar2">
    
        <include file="$(find rplidar_ros)/launch/rplidar_a2m8.launch" />
            <param  name="/rplidar2/rplidarNode/serial_port"  value="/dev/ttyUSB1"  />
    </group>

</launch>

YHuniv15, 2023-12-25:
Thank you.

What .launch file are used for the Rplidar 10N and 10 NP ? 
https://fr.aliexpress.com/item/1005004178345980.html?gatewayAdapt=glo2fra
This lidar is installed on this robot , but I get the error in scrren .
https://www.roboworks.net/store/p/rosbot-pro

I used :  for N10
<launch>
  <node name="rplidarNode"          pkg="rplidar_ros"  type="rplidarNode" output="screen">
  <param name="serial_port"         type="string" value="/dev/ttyUSB0"/>
  <param name="serial_baudrate"     type="int"    value="230400"/>
  <param name="frame_id"            type="string" value="laser"/>
  <param name="inverted"            type="bool"   value="false"/>
  <param name="angle_compensate"    type="bool"   value="true"/>
  </node>
</launch>

for N10P : 
<launch>
  <node name="rplidarNode"          pkg="rplidar_ros"  type="rplidarNode" output="screen">
  <param name="serial_port"         type="string" value="/dev/ttyUSB0"/>
  <param name="serial_baudrate"     type="int"    value="256000"/>
  <param name="frame_id"            type="string" value="laser"/>
  <param name="inverted"            type="bool"   value="false"/>
  <param name="angle_compensate"    type="bool"   value="true"/>
  </node>
</launch>

Thanks again

deyouslamtec, 2023-12-26:
N10/N10P is not our product, rplidar_ros does not adapt to this radar.You can consult the manufacturer.