Slamtec/rplidar_ros #143: RPLIDAR S2 UART (not using USB Adapter)

NeighborDva, 2023-11-06:
Hello.
I am a developer who is developing with S2 .

First i set UART Port on the expansion pin of the board called Beagle Board ai 64 and run RPLIDAR's ros package in a docker environment, 
the USB port (tyUSB) recognizes LiDAR and works well, but the UART port (ttyS) does not recognize LiDAR and does not run the driver.

I've disconnected the intermediate USB adapter, tried to connect directly, but the problem is still happening.

How do I use LiDAR with the expansion pin (UART) on the embedded board?

deyouslamtec, 2023-11-08:
We use the expansion of the Raspberry Pi 4B to connect Lidar to use lidar normally. It should be noted that the TX and RX pins need to be connected correctly to ensure that the Lidar power supply is sufficient.

NeighborDva, 2023-11-08:
> We use the expansion of the Raspberry Pi 4B to connect Lidar to use lidar normally. It should be noted that the TX and RX pins need to be connected correctly to ensure that the Lidar power supply is sufficient.

Then how do I use the pwm pin?

deyouslamtec, 2023-11-08:
Do not use this pin

closed 2023-11-28 by deyouslamtec