Slamtec/rplidar_ros #148: Error, unexpected error, code: 80008004

Vicenbs, 2024-01-03:
I'm trying to make work a Rplidar A1M8. I've go it connected to an adapter card that makes it work with usb. I'm using a StarFive VisionFive2 with Ubuntu 22.04 and ROS2 Humble. I keep getting this tipe of message:

`[INFO] [1704282704.950001785] [rplidar_node]: RPLidar running on ROS2 package rplidar_ros. RPLIDAR SDK Version:2.1.0`
`[ERROR] [1704282704.951693294] [rplidar_node]: Error, unexpected error, code: 80008004`
`[ros2run]: Process exited with failure 255`

Does anyone know what does this error mean and how to solve it?

hfcaio, 2024-01-08:
I'm having the same problem but in ros noetic. I'm using a rplidar s1, running in a docker container with ros noetic and ubuntu 20.04.

[image]

Did you find the solution ?

Vicenbs, 2024-01-09:
Not yet, a colleague of mine has told me it might be a driver problem, because I'm using a custom ubuntu image. In my case it doesn't detect the usbtty of the lidar, that might be your problem as well. Check it out just in case.

wette, 2024-01-09:
I had the same problem yesterday when I passed the usb device into docker like this:
`sudo docker run -it --rm -v /dev:/dev name_of_my_image bash`
This way, it is not possible to open the device from within the container.

But when I passed the device into the container using the --device option, the ros2 node worked flawlessly:
`sudo docker run -it --rm --device /dev/ttyUSB0:/dev/ttyUSB0  name_of_my_image bash`

Vicenbs, 2024-01-09:
@wette The thing for me is that I don't get the board to detect the ttyUSB.

wette, 2024-01-09:
@Vicenbs what is the output of dmsg when connecting the lidar to your computer? It may also be a power problem: does your USB port provide enough power to the lidar?

Vicenbs, 2024-01-09:
@wette Yes, i provide the power with the usb, but some colleagues of mine have done it this way with this same lidar and it worked for them, the actual problem seems to be the drivers, because as i'm working on an ubuntu image for risc-v it didn't come with them installed.

mateusguilhermedasilva, 2024-05-26:
> I'm having the same problem but in ros noetic. I'm using a rplidar s1, running in a docker container with ros noetic and ubuntu 20.04. [...]

Hi, I have the same problem. Did you manage to solve it?

wette, 2024-05-26:
If I remember correctly, It was a power issue.

The Lidar did not get enough power because the power supply I used for the raspberry pi was not powerful enough
--
Prof. Dr. rer. nat. Philip Wette
Hochschule Bielefeld - Campus Minden    Raum D 212
Artilleriestraße 9              Tel. +49.571.8385-211
32427 Minden                    ***@***.***

[image]<https://private-user-images.githubusercontent.com/83253047/294801169-6ebb3a1a-8c08-454a-bfbb-a422dfaba1f6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY2ODYyNDIsIm5iZiI6MTcxNjY4NTk0MiwicGF0aCI6Ii84MzI1MzA0Ny8yOTQ4MDExNjktNmViYjNhMWEtOGMwOC00NTRhLWJmYmItYTQyMmRmYWJhMWY2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI2VDAxMTIyMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTYxMWE4YzU4YWE5OThlOWUyNWFiMTg2MGUwMGRkNzVhMzFkZmRhYzVkNjdkZTYwOWJkODJkODQ5ZTBhMzJmNzUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.bWgvqIado16o1C8lMtCVd37LMUMLGGOzN7PeRMx5l94>

jiyoonhi, 2024-05-28:
I also have the same issue as follows
```
SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /rplidarNode/angle_compensate: True
 * /rplidarNode/frame_id: laser
 * /rplidarNode/inverted: False
 * /rplidarNode/serial_baudrate: 115200
 * /rplidarNode/serial_port: /dev/ttyUSB0

NODES
  /
    rplidarNode (rplidar_ros/rplidarNode)
    rplidarNodeClient (rplidar_ros/rplidarNodeClient)

auto-starting new master
process[master]: started with pid [65594]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f283ff5a-1d21-11ef-9652-ff8f8485fd25
process[rosout-1]: started with pid [65632]
started core service [/rosout]
process[rplidarNode-2]: started with pid [65639]
process[rplidarNodeClient-3]: started with pid [65640]
[ INFO] [1716921711.632260598]: RPLIDAR running on ROS package rplidar_ros, SDK Version:2.1.0
[ERROR] [1716921711.632835840]: Error, unexpected error, code: 80008004
[rplidarNode-2] process has died 
```
But I can see the device is detected.
```
[  948.329668] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[  962.629277] usb 3-1: new full-speed USB device number 13 using xhci_hcd
[  962.784212] usb 3-1: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[  962.784230] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  962.784234] usb 3-1: Product: CP2102N USB to UART Bridge Controller
[  962.787810] usb 3-1: cp210x converter now attached to ttyUSB0
```
```
$ ls -l /dev/ttyUSB0
crw-rw---- 1 root dialout 188, 0 May 28 13:44 /dev/ttyUSB0
```

Vicenbs, 2024-05-29:
> Hi, I have the same problem. Did you manage to solve it?

I used a rapsberry pi 3 to use the lidar and publish it to the ROS 2 environmet and it worked fine for me.

kimsooyoung, 2024-05-29:
This works for me, try out. 

```
cd <ros2-ws>/src
git clone -b ros2 https://github.com/Slamtec/rplidar_ros.git
cd ../
colcon build
source install/local_setup.bash 

cd /dev
sudo chmod 777 ttyUSB0

ros2 launch rplidar_ros <your-prefer-launch-file>
```

miyatat, 2024-06-18:
Me, as well.
Ref below:
https://github.com/robopeak/rplidar_ros/wiki

`Check the authority of rplidar's serial-port :

ls -l /dev |grep ttyUSB

Add the authority of write: (such as /dev/ttyUSB0)

sudo chmod 666 /dev/ttyUSB0

There're two ways to run rplidar ros package`