Slamtec/rplidar_ros #149: error, operation time out. SL_RESULT_OPERATION_TIMEOUT!

Junghyun124, 2024-01-09:
I'm using the Humble version for ros2.
I installed rplidar in the sbc workspace through the command below.

> sudo apt install ros-humble-rplidar-ros

In the foxy version, lidar worked well, but the error log appears as the operating time out error occurs, which I would have used by changing the version to humble.

I referred to it.
 [](https://github.com/Slamtec/rplidar_ros/issues/63)과 [](https://github.com/Slamtec/rplidar_ros/issues/5)

I've tried changing the cable here and I've tried changing the built rate. (I'm currently using 256000.)
I gave chmod 666 and 777 and tried using lidar nodes.
But it all comes up with an operation timeout. How do I fix it?

Below is the rplidar node code.
`import os
from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():

    return LaunchDescription([

        Node(
            package='rplidar_ros',
            executable='rplidar_composition',
            output='screen',
            parameters=[{
                'serial_port': '/dev/serial/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.3:1.0-port0',
                'frame_id': 'laser_frame',
                'angle_compensate': True,
                'scan_mode': 'Standard'
            }]
        )
    ])`

`robot@robot-desktop:~/robot_ws$ ros2 run rplidar_ros rplidar_composition --ros-args -p serial_port:=/dev/ttyUSB0 -p frame_id:=lidar_link -p angle_compensate:=true -p scan_mode:=Standard
[INFO] [1704785243.525071991] [rplidar_node]: RPLidar running on ROS2 package rplidar_ros. RPLIDAR SDK Version:2.0.0
[ERROR] [1704785247.587766150] [rplidar_node]: Error, operation time out. SL_RESULT_OPERATION_TIMEOUT! 
[ros2run]: Process exited with failure 255
robot@robot-desktop:~/robot_ws$ ls -l /dev/ttyUSB*
crw-rw-rw-+ 1 root dialout 188, 0 Jan  9 16:27 /dev/ttyUSB0
crw-rw----+ 1 root dialout 188, 1 Jan  9 16:28 /dev/ttyUSB1
robot@robot-desktop:~/robot_ws$ sudo chmod 666 /dev/ttyUSB0
[sudo] password for robot: 
robot@robot-desktop:~/robot_ws$ ros2 run rplidar_ros rplidar_composition --ros-args -p serial_port:=/dev/ttyUSB0 -p frame_id:=lidar_link -p angle_compensate:=true -p scan_mode:=Standard
[INFO] [1704785425.028944733] [rplidar_node]: RPLidar running on ROS2 package rplidar_ros. RPLIDAR SDK Version:2.0.0
[ERROR] [1704785429.084461004] [rplidar_node]: Error, operation time out. SL_RESULT_OPERATION_TIMEOUT! 
[ros2run]: Process exited with failure 255
`

Spirquel, 2024-01-24:
(my english bad, sorry) i have the same problem; my solution isn't in the rplidar_ros package. for me the problem is in the launch file to my package which use for launch my rplidar. in this package i have a rplidar_node.yaml this :
"/**:
  rplidar_node:
    ros__parameters:
      angle_compensate: true
      channel_type: serial
      flip_x_axis: false
      frame_id: laser_frame
      inverted: false
      scan_mode: ''
      serial_baudrate: 256000
      serial_port: /dev/ttyUSB0
      topic_name: scan
      use_sim_time: false
      
      # True turns off lidar motor when scan topic is not subscribed to. 
      # False runs lidar motor constantly. 
      auto_standby: true"

but my code search the "rplidar_composition" so i change in this file "rplidar_node" by "rplidar_composition".
i hope this comment help you; my package for launch my rplidar is here : https://github.com/iRobotEducation/create3_examples/tree/humble/create3_lidar_slam 
the proble is so the node execution and this name.

juandvergara, 2025-02-27:
> I'm using the Humble version for ros2. I installed rplidar in the sbc workspace through the command below.
> 
> > sudo apt install ros-humble-rplidar-ros
> 
> In the foxy version, lidar worked well, but the error log appears as the operating time out error occurs, which I would have used by changing the version to humble.
> 
> I referred to it. [#63](https://github.com/Slamtec/rplidar_ros/issues/63) [#5](https://github.com/Slamtec/rplidar_ros/issues/5)
> 
> I've tried changing the cable here and I've tried changing the built rate. (I'm currently using 256000.) I gave chmod 666 and 777 and tried using lidar nodes. But it all comes up with an operation timeout. How do I fix it?
> 
> Below is the rplidar node code. `
> 
> ```
>   import os
> 
>   from launch import LaunchDescription
> 
>   from launch_ros.actions import Node
> 
>   def generate_launch_description():
> 
>     return LaunchDescription([
> 
>     Node(
>         package='rplidar_ros',
>         executable='rplidar_composition',
>         output='screen',
>         parameters=[{
>             'serial_port': '/dev/serial/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.3:1.0-port0',
>             'frame_id': 'laser_frame',
>             'angle_compensate': True,
>             'scan_mode': 'Standard'
>         }]
>     )
> ])`
> ```
> 
> > ros2 run rplidar_ros rplidar_composition --ros-args -p serial_port:=/dev/ttyUSB0 -p frame_id:=lidar_link -p angle_compensate:=true -p scan_mode:=Standard
> 
> `[INFO] [1704785243.525071991] [rplidar_node]: RPLidar running on ROS2 package rplidar_ros. RPLIDAR SDK Version:2.0.0 [ERROR] [1704785247.587766150] [rplidar_node]: Error, operation time out. SL_RESULT_OPERATION_TIMEOUT! [ros2run]: Process exited with failure 255 robot@robot-desktop:~/robot_ws$ ls -l /dev/ttyUSB* crw-rw-rw-+ 1 root dialout 188, 0 Jan 9 16:27 /dev/ttyUSB0 crw-rw----+ 1 root dialout 188, 1 Jan 9 16:28 /dev/ttyUSB1 robot@robot-desktop:~/robot_ws$ sudo chmod 666 /dev/ttyUSB0 [sudo] password for robot: robot@robot-desktop:~/robot_ws$ ros2 run rplidar_ros rplidar_composition --ros-args -p serial_port:=/dev/ttyUSB0 -p frame_id:=lidar_link -p angle_compensate:=true -p scan_mode:=Standard [INFO] [1704785425.028944733] [rplidar_node]: RPLidar running on ROS2 package rplidar_ros. RPLIDAR SDK Version:2.0.0 [ERROR] [1704785429.084461004] [rplidar_node]: Error, operation time out. SL_RESULT_OPERATION_TIMEOUT! [ros2run]: Process exited with failure 255
> 
> `
> 
> [image]
> 
> [image]

to solve that problem add in launch file the parameter 'serial_baudrate' to 115200. by default is 100000. And check is the usb port is correct, usually is ttyUSB0 o ttyUSB1.