Slamtec/rplidar_ros #152: S2 - Error, RPLidar internal error detected. Please reboot the device to retry

CycleMark, 2024-02-03:
Re the S2 - There seems to be various "ideas" on how to solve this. Including SLAMTec's very own "Unplug it and plug it back in again nonsense.

This sensor has never worked. Latest drivers, good PSUs and UBS cables.

@slamtec - what gives? 

Mark

oguzhankose, 2024-05-27:
Hello Mark
i agree that unplugging and plugging it back in is not a way for solving a problem. But when we talk to rplidar about this problem, they couldn't give any other recommendation to us. So i think there is no solution for this problem for now. If you find a solution, please let me know.

3bdul1ah, 2024-09-23:
I'm currently facing same issue with my RPLidar S2 on the Jetson Nano. The lidar works perfectly fine on my laptop, but when I use it with the Jetson Nano, I consistently get the following error:

Error: rplidar internal error detected. Please reboot the device to retry.

I've tried the following solutions:

- Granted proper permissions (chmod for /dev/ttyUSB0)
- Ensured proper wiring, including separate power supply
- Rebooting the device and reconnecting multiple times
- Running sudo apt update && sudo apt upgrade
- Plugging and unplugging the device
When I unplug and reconnect, it works for about 2 seconds and then fails with the same internal error.

This seems to be a common problem with the S2 lidar, as many users have reported similar issues. Has anyone managed to figure out the exact reason for this behavior or found a reliable solution?

Robotic01, 2024-10-31:
How often did you get the problem? I've got the problem too

3bdul1ah, 2024-11-01:
> How often did you get the problem? I've got the problem too

I managed to solve the issue. I was using a Jetson Nano with Ubuntu 20.04, which isn’t natively supported as the Jetson Nano is optimized for Ubuntu 18.04. This mismatch likely caused some limitations, such as the USB port not supplying enough voltage to power the LiDAR, which could be why we kept seeing the error.

After downgrading to Ubuntu 18.04 and testing it, surprisingly, everything worked without any issues. From my experience, this seems to be a reasonable explanation. However, the error is quite general and could occur for different reasons, making it hard to pinpoint the exact cause.

While it’s working fine on the Jetson Nano with Ubuntu 18.04, I did notice that the error could also stem from changes in the device path, like /dev/ttyUSB*, which keeps switching between 1 and 2. This doesn’t seem like an issue that can be permanently fixed by assigning it to a single port. So, even after downgrading (which worked for me), you might still encounter the error, but likely due to the port switching. In that case, it should get back to working fine once you adjust it.

CycleMark, 2024-11-15:
Seems to be like the device is still broken. 18.04 is very old.

A-Nobel, 2024-11-30:
I found that by quickly inserting the microUSB cable and power cord within 2-3 seconds, it works.

CycleMark, 2024-11-30:
Unfortunately not a solution. The device cannot have manual intervention to make it "work"

Titan43, 2025-07-01:
Also encountered this issue. I've tried to send a RESET signal once per few hours and such clanky solutions seems to work