Slamtec/rplidar_ros #164: Updating release repo for newer ROS distros

joshnewans, 2025-05-20:
In the ROS package repositories, this repo is the source for the rplidar-ros package on the older distro humble.

In newer distros (e.g. jazzy, kilted, rolling) this package is built from https://github.com/allenh1/rplidar_ros which appears to be older and more out-of-date than this one.

@deyouslamtec you are listed as the maintainer for Humble, as an official representative can you please get the packages for newer distros updated to your repo, especially jazzy (current LTS) and rolling (will set the new standard).

Thanks,
Josh

bjsowa, 2025-07-11:
@deyouslamtec Could you please look into that?

anilcemalyasar, 2025-08-03:
@deyouslamtec Could you please look into that?